bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #7: 3D Step file incomplete & errors.

NAPCAL, 2021-05-23:
Step issues:
BLtouch connector has no pins.
[image]

5V-V_bus header; only one pin showing, and it is not through the board.
[image]

Also, the Daig & Boot0 connection headers are missing a pin and not through the board. 
[image]
[image]

Power inputs and Bed output terminals are not taking up space on the board; it should start near F1 and be down to near R63.
[image]
[image]

Heater connections; only showing two sets of terminals when it should be four sets.
[image]
[image]

thisiskeithb, 2021-05-25:
Fixed in https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/commit/ec4a590edc5a7125455eeadd3bc3961491575fb9:

[image]

MakerJinn, 2021-06-15:
hi, I couldn't download the 3D step file listed. when I try to download it, it will open another window with codes like this
(i assume its the geometry of the parts)

how can I download this file? 
Thanks a lot.

DATA;
#10=CURVE_STYLE('',#272,POSITIVE_LENGTH_MEASURE(0.1),#15793);
#11=CURVE_STYLE('',#273,POSITIVE_LENGTH_MEASURE(0.1),#17518);
#12=CURVE_STYLE('',#274,POSITIVE_LENGTH_MEASURE(0.1),#17519);
#13=CURVE_STYLE('',#275,POSITIVE_LENGTH_MEASURE(0.1),#17520);
#14=CURVE_STYLE('',#276,POSITIVE_LENGTH_MEASURE(0.1),#17521);
#15=CURVE_STYLE('',#277,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#16=CURVE_STYLE('',#278,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#17=CURVE_STYLE('',#279,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#18=CURVE_STYLE('',#280,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#19=CURVE_STYLE('',#281,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#20=CURVE_STYLE('',#282,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#21=CURVE_STYLE('',#283,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#22=CURVE_STYLE('',#284,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#23=CURVE_STYLE('',#285,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#24=CURVE_STYLE('',#286,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#25=CURVE_STYLE('',#287,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#26=CURVE_STYLE('',#288,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#27=CURVE_STYLE('',#289,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#28=CURVE_STYLE('',#290,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#29=CURVE_STYLE('',#291,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#30=CURVE_STYLE('',#292,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#31=CURVE_STYLE('',#293,POSITIVE_LENGTH_MEASURE(0.1),#42384);
#32=CURVE_STYLE('',#294,POSITIVE_LENGTH_MEASURE(0.1),#42384);

tregnier, 2021-06-22:
> hi, I couldn't download the 3D step file listed. when I try to download it, it will open another window with codes like this
> (i assume its the geometry of the parts)
>
> how can I download this file? [...]

I use github desktop to download step file. There is way to do it through web interface directly, but much easier with github desktop

Pixeltrace-GER, 2021-07-16:
[image]

Just press the Download Button or clone the Repo