bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #22: Octopus not detected

sims-60, 2021-06-08:
I connected the octopus to my pc and then on the octopus the leds also light up but the pc don't detect the octopus board

dazkit, 2021-06-08:
try that https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/17

sims-60, 2021-06-08:
ok

dazkit, 2021-06-08:
[image]
you have to put this jumper to connect via usb

sims-60, 2021-06-08:
but when it is the sd card reader why the pc then can't connect to the board

dazkit, 2021-06-08:
do you use an external power supply or do you only connect it with usb? If you only connect it with USB you have to set the jumper next to the Power LED

sims-60, 2021-06-08:
i use usb and i have set the jumper and the leds also light up

dazkit, 2021-06-08:
Try to set the Jumper at Boot0. Did  you get a USB connection?

sims-60, 2021-06-08:
yes

dazkit, 2021-06-08:
I had the same Problem after i installed the firmware.bin then i get a connection. Its not a Problem from the SD Card reader. Try to reinstall the firmware.bin

sims-60, 2021-06-08:
what do you mean with reinstall the firmware.bin

dazkit, 2021-06-08:
Firmware is marlin or klipper. Which one do you want to use?

sims-60, 2021-06-08:
klipper

dazkit, 2021-06-08:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/main/BIGTREETECH_Octopus_EN.pdf

here on page 20. You have to create a firmware.bin and insert it into the board with a MicroSD card and transfer it

sims-60, 2021-06-08:
i know

sims-60, 2021-06-08:
have you discord? because i cant send my firmware on github github not support .bin files

dazkit, 2021-06-08:
sorry i have no discord. I had the same Problem after i create the firmware.bin and transfer it with a sd card my board was not recognized with USB. Than i used a different SD Card and now it works. I think you have the Same Problem as me. If that's not the problem, then your board is probably broken

sims-60, 2021-06-08:
do you use klipper?

sims-60, 2021-06-08:
and  i have the exat same problem

sims-60, 2021-06-08:
can you send me your firmware.bin?

sims-60, 2021-06-08:
here is my email

sims-60, 2021-06-08:
fkp00065@zwoho.com
its a trash mail which only exist for 10 minutes

dazkit, 2021-06-08:
okay

sims-60, 2021-06-08:
I WORKS

sims-60, 2021-06-08:
i think i had an error im my firmware

dazkit, 2021-06-08:
Ok then close this thread and have fun :)

closed 2021-06-08 by sims-60

lowtolerance, 2021-06-13:
Can I have this magic firmware.bin? I can't seem to get this board to recognize my binary off of the SD card, and IDK if it's the binary of the board. A known-good klipper binary would be a big help, and I'm not sure why BTT doesn't give publish one themselves.

sims-60, 2021-06-13:
hi  can you send a mail to this mail : sqf86920@eoopy.com?
then i will send you the firmware

sims-60, 2021-06-13:
because i can't send a .bin file with github

lowtolerance, 2021-06-13:
I actually managed to sort this out myself, I don't think your firmware
would have helped. I jumped two of the wrong pins for sensorless homing and
I think it was putting the bootloader into an error state.

sims-60, 2021-06-13:
oh ok in this moment i wanted to send you the email XDDDD

Voodoobrew101, 2021-09-30:
I dont use klipper, but in Marlin these sound like the types of issues you get when you use the wrong 
#define SERIAL_PORT           1  // TFT
#define SERIAL_PORT_2      -1  // USB
#define SERIAL_PORT_3       2  // RPI, Raspberry Pi
settings. In my experience the order is also important. My settings are as above, which are the defaults in this github.

drelich, 2022-09-07:
I know this is a year old topic, and I hate to resurrect it, but just was dealing with the same thing. For two hours I was pulling my hair because I couldn't see any response to either `lsusb` nor `ls /dev/serial/by-id/*`

Turned out I accidentally compiled Klipper FW with the wrong MCU frequency. My board is model 429 with MCU freq. of 8MHz but I compiled the FW for the other version with 12MHz. Once I realised my mistake and recompiled FW with the correct frequency, the board indeed came back to life and I could see meaningful output to both the `lsusb` and `ls /dev/serial/by-id/*` commands.

Just thought I'd share in case somebody made the same mistake :)

whatder1, 2025-01-01:
> I know this is a year old topic, and I hate to resurrect it, but just was dealing with the same thing. For two hours I was pulling my hair because I couldn't see any response to either `lsusb` nor [...]

How did you get the updated firmware installed?  Using the SD?  I ask because I think I may have bricked mine but I can't seem to update the firmware using the SD card (downloaded a stock firmware file from BTT's GitHub, renamed it to firmware.bin and inserted into the Octopus).  the firmware file hasn't been renamed so I suspect that it isn't flashing...