bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #24: Klipper UART

DJames2682, 2021-06-10:
It appears the uart pins for 2208's are labled as expansion pins. 
Is this correct ?
for example PE8 is labled as Motor uart pin but also labled as an expansion pin EXP1_1.

dazkit, 2021-06-11:
You have to adjust it. These are the uart pins. The ones marked in red. For uart you only need the #uart_pin. You can delete the #tx_pin reference.

this is from the new Pin.pdf
[image]

tregnier, 2021-06-22:
that worked for me like this too.

Don't follow the config file for klipper configuration from this github, just set the jumper as shown on the manual, and then set UART_pin based on CS column mention here. nothing else needed
Thanks dazkit