bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #28: Unknown board ID 'marlin_BigTree_octopus_v1'

user-e, 2021-06-20:
How to reproduce:

- Clone the rep
- Open it with Vs Code with PlatformIO
- Build

Result:
```

Executing task: platformio run

Processing BIGTREE_OCTOPUS_V1_0 (platform: ststm32@~12.1; board: marlin_BigTree_octopus_v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'marlin_BigTree_octopus_v1'
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Default Configuration.h, default Configuration_adv.h, and default Platformio.ini

DJames2682, 2021-06-20:
Hey I just did this. Go into the boards.h file in marlin 2.0.9 highlight
the Btt_ octopus and copy the info that pops up. Exactly.

They changed the name. In 2.0.9

user-e, 2021-06-20:
@DJames2682 
I'm not sure to had understood you. That's the popup that appear in boards.h file at the octopus declaration:

[Image](https://i.postimg.cc/TP6YDrG5/Schermata-da-2021-06-20-20-25-31.png)

bwspath, 2021-06-23:
the board name in ini/stm32f4.ini is wrong:
` -board             = marlin_BigTree_octopus_v1`
vs
`+board             = marlin_BigTree_Octopus_v1`
that  should fix it.

DJames2682, 2021-06-23:
Sorry I went back and looked. The build environment that worked or me was BTT_OCTOPUS_V1 in 2.0.9.

Also. 

Octopus 1.0. thermal runaway and connection issues after bootloader and firmware update.
Almost burnt it down . Cant communicate with the pc to reload the new bootloader, flash firmware, and the mcu get so hot it resets every 15 sec. Oh and every heater turns on and stay on. Be careful.

user-e, 2021-06-23:
> the board name in ini/stm32f4.ini is wrong:
> ` -board = marlin_BigTree_octopus_v1`
> vs
> `+board = marlin_BigTree_Octopus_v1`
> that should fix it.

It works! Perfect!

closed 2021-06-23 by user-e