bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #31: Urgent issue

DJames2682, 2021-06-22:
Upon loading the new bootloader, USB Counication will not establis, 
New firmware.bin files will not flash. The board constantly reboots and the MCU is extreamly hot. 

More importantly. 
A major issue with thermal runaway. All heaters on the board when recieving power for the main begin to heat uncontrollably. 
this occurs with the Bed, Heater 0 and 1 (I note that all the indicator lights for the heaters are on. 

This could have lead to a fire or metdown. 
I've seen several other posets in varrious forums to this effect.

user-e, 2021-06-23:
@DJames2682 
I know that this is not a forum but...
If you are no more able to flash the MCU through DFU or SD card, you can try to program it with the ST-Link Programmer and write the older bootloader.

DJames2682, 2021-06-23:
It is not but I thought the situation with the thermal runaway warented someone looking into it. 
The board will connect to the pc but the software will not recognize it.

looxonline, 2021-07-03:
As far as I understand this was because of some incorrect wiring on the probe. Is that correct?

DJames2682, 2021-07-03:
Yep. Had the pins switches on the z endstop w/ a bltouch....not entirely
sure why I caused the heater issues