bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #33: Pin PB7 on J40 is not working

ewalbridge, 2021-06-25:
Pin PB7 on J40 (SENSOR) is not working for the inductive proximity sensor but when I connect it to PB7 J43 (BL_TOUCH) it works. The inductive proximity sensor is 24v so I need the 24v from J40 and not the 5v from J43.

Is there a setting I am missing or a jumper that I need to set to enable PB7 to work on J40?

looxonline, 2021-07-03:
There are a few options available to get this to work. You likely have an NPN style sensor which means that you will need to add an external pull up. To you this you just need to connect a 4k7 resistor between the voltage pin of the probe and the signal pin. Then you will need to configure your firmware to invert the input signal and enable the pull down, ~! in Klipper. There will be no BAT85 required with this config.

denisrohr, 2021-07-12:
If you look at https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/Hardware/BIGTREETECH%20Octopus.pdf
you can see the pin "PROBE" on J40 is attached to an optocoupler EL357C with a 1k resistor. The optocoupler has a 1.4V forward voltage and a max current of 50mA. This means you can attach any signal to it directly without any diode, even 48V should be fine. Attach the 12V-48V signal of the probe directly to the third pin of J40. The third pin is NOT PB7, but connects to an optocoupler which sets PB7 to HIGH if PROBE has atleast 12V voltage. 
DO NOT attach PROBE directly to PB7 on J43.

I use a NO inductive probe without any diodes at 12V supply voltage and connected the signal pin directly to PROBE on J40.
My klipper config is:

...
[probe]
pin: ~PB7
...

ga73, 2021-07-17:
This really needs to be resolved and poor by BTT to just not answer... Doesn't give me any reason to buy their products when others such as FYSETC actually respond to issues!

Answer please with a solution BTT!

looxonline, 2021-07-17:
> This really needs to be resolved and poor by BTT to just not answer... Doesn't give me any reason to buy their products when others such as FYSETC actually respond to issues!
> 
> Answer please with a solution BTT!

There is no point for them to respond when the answer has already been given in this thread. If you took the time to read it instead of getting aggressive and threatening then you would already have your issue solved. Otherwise, there is always the Fysetc which will literally kill your LCD in a puff of smoke and a high % of the boards were suiciding on their first power up with drivers installed.

looxonline, 2021-07-17:
> If you look at https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/Hardware/BIGTREETECH%20Octopus.pdf
> you can see the pin "PROBE" on J40 is attached to an optocoupler EL357C with a [...]

This is only true if your probe is PNP style. I.e. that it will output a high voltage when triggered (or when not triggered) and simply float when in the opposite state. If you have an NPN style probe that pulls to ground when active and floats otherwise then you'll need to add a little pull up resistor between the probe signal pin and the voltage supply to the probe. That will provide the current source for when the probe leaves the signal pin floating that will in turn drive the optocoupler.

ga73, 2021-07-17:
> There is no point for them to respond when the answer has already been given in this thread. If you took the time to read it instead of getting aggressive and threatening then you would already have [...]

Nobody is "getting aggressive", I'm looking for an official reply from btt who seem to be happy to let their customers sort out their problems, there are 2 answers both of which aren't clear to me and I'm guessing many others who aren't as electronically minded as some.

Your reply seems a little bit pointless as it's of absolutely no benefit, maybe if you're going to reply make your replies helpful or just say nothing.

looxonline, 2021-07-17:
> Nobody is "getting aggressive", I'm looking for an official reply from btt who seem to be happy to let their customers sort out their problems, there are 2 answers both of which aren't clear to me [...]

I encourage you to browse through these github issues. You should notice that BTT don't use this as a channel to provide customers with support. Their github is a repo to provide customers with info on their products. They have specific support channels which can be found on their discord server, facebook groups and websites. The issues tab is left open specifically for customers to discuss things in an open forum. Granted, it may be best for them to include that info in their readme files to make it clear.

My reply is quite helpful actually, maybe not to you but to others. Why? 

1.) It should help others to see that there is actually no problem whatsoever with this input. Users simply need to connect up to it in the right way. 

2.) It should help users who read your response demanding a reply from btt to clearly understand that the answer has already been provided. Twice on this thread in fact. They therefore can take the info that has already been provided and apply it instead of being left in doubt as to whether or not it will work.

3.) It should help users to be aware of the issues with the Fysetc board so that they can make an educated decision about whether or not they would be willing to take the chance with it.

Back to the point at hand. To fix your connection you need to:

> "connect a 4k7 resistor between the voltage pin of the probe and the signal pin."

In addition to the above you need to remove the BAT diode completely as it is not needed with this input. It is an optocoupler which is better than the standard inputs on other boards and has no need for a protection diode. Instead all it needs is a pull up resistor with NPN style probes which is evidently what you are using.

ga73, 2021-07-17:
> I encourage you to browse through these github issues. You should notice that BTT don't use this as a channel to provide customers with support. Their github is a repo to provide customers with info [...]

Hi thanks for the reply which is helpful.

So in laymens terms, I simply need to solder the 4k7 resistor to both the v+ and the probe sensor wire and remove the bat85 resistor?

Many thanks
Ga

looxonline, 2021-07-17:
> Hi thanks for the reply which is helpful.
>
> So in laymens terms, I simply need to solder the 4k7 resistor to both the v+ and the probe sensor wire and remove the bat85 resistor?
>
> Many thanks
> Ga

Exactly. Solder or crimp into the connector terminals for the probe. Whichever works best for you.

ga73, 2021-07-17:
Thank you very much! 😊👍

mj2068, 2021-07-18:
i've been combating this issue for a whole day in the office, on a Sunday. And now, it's 1 am in the morning, i'm still reading on the web trying to find answers to prepare for tomorrow. i'm so so happy i found this post, and it seems exactly like my situation, so, after reading the whole thing, i'm fairly confident, it'll be solved right after i shovel that 4.7k R in there, first thing in the morning.
people like you, @looxonline, are the unsung heroes of the internet, for that, i salute to you.

onurddd, 2021-08-04:
Can anyone share marlin config file with working probe settings please

looxonline, 2021-08-04:
> Can anyone share marlin config file with working probe settings please

What kind of a probe are you using and where have you connected it. There is not just one config file because there are hundreds of different probe permutations out there.

onurddd, 2021-08-04:
Sorry, I use NPN type capacitive sensor that connected to j40 header. I will not use  z min switch . Capacitive probe will be  used for bed leveling and  as a z min limit switch.

looxonline, 2021-08-04:
> Sorry, I use NPN type capacitive sensor that connected to j40 header. I will not use z min switch . Capacitive probe will be used for bed leveling and as a z min limit switch.

Then you will need to apply the pull up resistor mentioned earlier in this thread. You will also need to enable a software pull down on PD7 within Marlin and likely set the signal to inverting for the probe on PD7.

looxonline, 2021-08-04:
> > > Sorry, I use NPN type capacitive sensor that connected to j40 header. I will not use z min switch . Capacitive probe will be used for bed leveling and as a z min limit switch.
> > 
> > 
> > Then you will need to apply the pull up resistor mentioned earlier in this thread. You will also need to enable a software pull down on PD7 within Marlin and likely set the signal to inverting for the probe on PD7.
> 
> I tried all of them and it doesnt work. That's why I need a working config file. If you have one it is much appreciated.

Unfortunately I don't but I know that it does work from having helped many others to get it to work. You may also need to check that the pin mappings in the pins file are correct too. Also remember that you can't/don't need to use a BAT85 with this board.

Teila, 2021-08-05:
While I am not using the octopus on my marlin controlled system (klipper based  for this guy) I had to use the following setup for my inductive probe.

[probe]
pin: ~PB7 #enable micro-controller pulldown

amazon.com/gp/product/B08MX6SHD4 <- Probe I am using

Aurawin, 2021-08-07:
I've just got mine working with no resistor by re-mapping the J31 diag pin PG10 -> you can re-map marlin to look there for the low voltage signal - assuming your inductive probe goes to ground when triggered.

#define Z_MIN_PROBE_PIN PG10
#define Z_STOP_PIN PG10

onurddd, 2021-08-09:
> While I am not using the octopus on my marlin controlled system (klipper based for this guy) I had to use the following setup for my inductive probe.
> 
> [probe]
> pin: ~PB7 #enable micro-controller pulldown
> 
> amazon.com/gp/product/B08MX6SHD4 <- Probe I am using

As Teila said, Bp7 pin needs to be pulled down. İf you look at the octopus pins you can see Probe has a optocoupler at j40 header. Then this optocoupler connected to Pb7 pin . There is an exposel Pb7 pin on j43 header if you measure  the pb7 and gnd pins with a voltmeter, you need to measure 3.3 volts. İf you trigger the probe (j40 header)  without any pull down and pull up pb7 voltage goes to 0.6-0.7 volts and signal doesn't count as 0. If you assign pull down to Pb7, when you trigger the probe Pb7 voltage goes to 0.08 and it counts as 0 signal.

Aurawin, 2021-09-22:
I wanted to try using the optocoupler.  This time I'm using a NO PR18-5DP Cylindrical Type Detector 5mm DC12-24V 3-Wire PNP NO(Normally Open) with 24V selected input wired to J40.  If I measure PB7 on J43 when the sensor is triggered I can see the 3.3V when triggered and 0.4-.7v when not triggered.  

#define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1

// No combination of Endstop pulling changes the fact that marlin will not detect the probe
// #define ENDSTOPPULLUP_ZMIN_PROBE
// #define ENDSTOPPULLDOWN_ZMIN_PROBE

#define Z_MIN_PROBE_PIN PB7

But no matter what I do I can't get Marlin to detect the signal using M119 - It's always triggered for Non-Inverted signals or always open for signal inversion.

I'm using 2.0.9.1 from Marlin direct, are there any changes lately that the upstream repo doesn't have regarding this issue?

looxonline, 2021-09-22:
> I wanted to try using the optocoupler. This time I'm using a NO PR18-5DP Cylindrical Type Detector 5mm DC12-24V 3-Wire PNP NO(Normally Open) with 24V selected input wired to J40. If I measure PB7 on [...]

Try the latest bugfix. I've had issues with 2.0.9.1.

onurddd, 2021-09-22:
> I wanted to try using the optocoupler. This time I'm using a NO PR18-5DP Cylindrical Type Detector 5mm DC12-24V 3-Wire PNP NO(Normally Open) with 24V selected input wired to J40. If I measure PB7 on [...]

Try to cancell all the pull ups and activate pulldown for  Z_MIN_PROBE_PIN PB7, ZMIN_PROBE and ZMIN. I couldn't make it work untill the pull down resistors are activated.

Aurawin, 2021-09-22:
@onurddd , I tried enabling the pulldowns but got this error on building
Compiling .pio\build\BIGTREE_OCTOPUS_V1\src\src\HAL\STM32\eeprom_sdcard.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:28:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:725:4: error: #error "PULLDOWN pin mode is not available on the selected board."

Aurawin, 2021-09-23:
Update:  Got it working.  Looked in Marlin Source code and the pulldown code wasn't being executed.  The reason being
Configuration.h ...

#define USE_ZMIN_PLUG
#define USES_Z_MIN_PROBE_PIN PB7

These 2 defines need to be declared in order for the ENDSTOPPULLDOWN_ZMIN_PROBE to pull the voltage down on the optocoupler.

rakeshdutta90, 2021-09-30:
> This is only true if your probe is PNP style. I.e. that it will output a high voltage when triggered (or when not triggered) and simply float when in the opposite state. If you have an NPN style [...]

Does that mean you are saying if I use PNP style probe and select 12v/24v from J38 and connect the signal wire from the probe directly to PB7 (J40) will get the reading from without any issue?

rakeshdutta90, 2021-09-30:
> Exactly. Solder or crimp into the connector terminals for the probe. Whichever works best for you.

So if I use PNP NO (https://www.amazon.in/gp/product/B07RCFBXRB/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 ) inductive sensor then just plug and play nothing I have to do?

jsmith59, 2022-01-26:
I had the same/similar problem using an NPN probe (prusa superpinda). Initially Z homing acted like the probe signal was inverted, but when I inverted the signal, it wouldn't trigger.  Running Marlin 2.0.9 bug fix. I had already added the 4k7 resistor. Through the suggestions in the thread, here is what finally worked for me:

#define USE_ZMIN_PLUG (enabled by default)
#define Z_MIN_PROBE_PIN PB7 (may not be necessary since this is the default in the pins file for this board

**I had to disable ENDSTOPPULLUP_ZMIN_PROBE:**
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_IMIN
  //#define ENDSTOPPULLUP_JMIN
  //#define ENDSTOPPULLUP_KMIN
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_IMAX
  //#define ENDSTOPPULLUP_JMAX
  //#define ENDSTOPPULLUP_KMAX
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

**I had to enable ENDSTOPPULLDOWN_ZMIN_PROBE:**
// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_IMIN
  //#define ENDSTOPPULLDOWN_JMIN
  //#define ENDSTOPPULLDOWN_KMIN
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_IMAX
  //#define ENDSTOPPULLDOWN_JMAX
  //#define ENDSTOPPULLDOWN_KMAX
  #define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif

**I left Z_MIN_PROBE_ENDSTOP_INVERTING false.**