bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #35: Wifi Module ? wich one it's working with RRF ?

valenti1234, 2021-07-03:
Can you please specify witch WIFI module need to be used with the octopus ?

thisiskeithb, 2021-07-04:
[BigTreeTech ESP8266 WiFi Module](https://www.bigtree-tech.com/products/bigtreetech-esp8266-wifi-module.html)

valenti1234, 2021-07-04:
> [BigTreeTech ESP8266 WiFi Module](https://www.bigtree-tech.com/products/bigtreetech-esp8266-wifi-module.html)

no way to redefine board.txt in order to work with : [](https://www.bigtree-tech.com/products/bigtreetech-rrf-wifi-v1-0.html) , I have a bunch of them and also faster to order in Amazon

Moorviper, 2021-08-16:
On their store on aliexpress the Sell the ESP-07S as matching for the octopus.
when setting debug mode on it lists my wifi networks.
But i can't connect to them.
I also can set it up as an access point.

Moorviper, 2021-08-30:
I now tried both the BTT S07 and BTT S12 module.
None of them work.
Both can scan but you can't add the wifi or setup an access point with them.