bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #40: Request source code for RRF changes

matthewlloyd, 2021-07-13:
Hello,

I own an Octopus board and was very happy to discover RRF has been ported to the board. I found the firmware binary in this repo, but no accompanying source code.

RRF is licensed under GPLv3. Could you please provide the source code for the changes you made to RRF to get it to compile and fit?

Many thanks,
Matthew

valenti1234, 2021-07-13:
Same here source code are due ! we need variants files in order to port / fix the firmware

bigtreetech, 2021-07-15:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/tree/master/Firmware/RepRapFirmware/RRFBuild-SourceCode
Hello, here is the source code

matthewlloyd, 2021-07-15:
Absolutely fantastic, thank you so much, especially for doing this so quickly! I have great respect for BTT's products and service (and my girlfriend loves the rubber duckies :) ).

closed 2021-07-15 by matthewlloyd

valenti1234, 2021-07-17:
RRF source build, but look like the generate SBC and ESP  firmware dont work anyone success it ?