bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #41: max output for neopixel port ?

DocYester, 2021-07-13:
what amount of led's can be connected to the neopixel-port ?
or what is the max output in mA on PB0 (J37) ?

Seminole81, 2021-08-12:
Did you find out? I am also wondering this.

DocYester, 2021-08-16:
No, i'll keep it as for other boards ... 30 led max -> 1,5A peak
Actually running it without problems

Seminole81, 2021-08-16:
I read somewhere last night that the max. amps this board can handle is 8. However, I have heard from BTT that it can do up to 30 without max. current. I saw someone 44 fine.

DocYester, 2021-08-16:
8A all over the 5V rails ... so you cant use it all for calculating for led's - i read it in the docs too
i think it could handle several more, but i wont test it just to fry the board

so i'm waiting for an official answer

Seminole81, 2021-08-18:
Ok, well someone else I know on Facebook is successfully running 60 LEDs with the board. BTT told me 30 via an email (guessing that's a safe one).

matthewdolman, 2021-11-03:
Oops, I didn't realise this could fry the board, I just thought it would stop working if too much power gets drawn.

I ran 144, with nothing else going, I tried to add another 72 but got a message in klipper that the max was 166.  It would be nice if there was some documentation or even a response from BTT on this.

simplified-life, 2022-12-08:
I'm just about to wire up 90 neopixel LED's, but thinking perhaps it is better to power them all from with a 24v to 5v step down and then control them with the Raspberry Pi. Or should I control them off the Bigtreetech Octopus Pro 1.0 board I've got coming?

matthewdolman, 2022-12-09:
You will need a 5v power supply that can handle the current of the LEDs the normal budget for neopixels is 0.05A so for 100 you will want 5A from your power supply, in addition you will need power for your raspberry pi.  I ended up getting a 16A one to give me plenty of room.