bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #42: PROBE SENSOR NOT WORKING

ga73, 2021-07-17:
Can someone advise on how to get the probe sensor working, its all wired correctly and has the bat85 installed.. Probe lights up and off when putting something metal underneath but is not sensed in the fw. 

Does this need moving to another pin or ??

looxonline, 2021-07-17:
> Can someone advise on how to get the probe sensor working, its all wired correctly and has the bat85 installed.. Probe lights up and off when putting something metal underneath but is not sensed in the fw.
> 
> Does this need moving to another pin or ??

The answer to this has already been provided in the other issue where you had a rant. Just read it and apply it and you will be good to go.

Aurawin, 2021-08-07:
Alternatively, I've just got mine working with no resistor by re-mapping the J31 diag pin PG10 -> you can re-map marlin to look there for the low voltage signal - assuming your inductive probe goes to ground when triggered.

#define Z_MIN_PROBE_PIN PG10
#define Z_STOP_PIN PG10

Just remove the actual pin from the 3 pin JST connector and manually put it into PG10.

Hman32, 2022-04-16:
Will be dangerous if the probe does not output 5v or less when he trigger...
The sensor port is intended to be protected with a optocoupler, just sad is not working as expected.
In my case I ended adding an external optocoupler and I've used the pin for the Z endstop.