bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #45: Board not detected by Windows

PrideParrot, 2021-07-24:
Hi, 
just got my Octopus 1.1 and try to upload Marlin on it. But after connecting board via usb device not detected by Windows (no new devices in Device Manager, any type), aslo not detected by STM32CubeProgrammer (any connection type). Tryed to power board via usb-c or via external 24v supply (without J68 jumper), but no differents.  If I understand correctly, I don't have bootloader on board. But uploading it via SD-card not working (no “FIRMWARE.CUR” on SD-card in the end).  

Is it possible to alive board with USBasp or else some how?

Windows: 10 Home
Version: 20H2
OS: 19042.1110

DJames2682, 2021-07-25:
Try removing all the other connections ie stepper an fan cables etc.  That's what I had to do m theory is t was trying to feed power to the components with connected.

PrideParrot, 2021-07-25:
@DJames2682 , all discriben earlie was tested without any additional load on board. Also USB-C connection was tested via USB-C to USB-C wire and I get the same result as USB-C to USB-A that was in the box with board.

DJames2682, 2021-07-25:
It took a little persuasion to get mine to connect. On a windows computer I went under device manager and deleted all of the com ports. Took the board completely out of the printer. Installed the jumpers and uses an actual USB 3.0 cable . Had the software running when I reconnected the board. I'm not sure what of this combination worked for me but it was a little touchy getting it to connect to the pc via the bootloader software.

PrideParrot, 2021-07-28:
Cheched variant with smaller SD card - with no result.

Tryed to connect via St-Link v2 (PA13(SWDIO), PA14(SWCLK)) but St-Link don't found target (STM32 ST-LINK Utility and STM32CubeProgrammer). Seems my board is a brick.

DJames2682, 2021-07-28:
Try dissconnecting everything from the board. Worked for me

PrideParrot, 2021-07-28:
All described earlier was tested only with board without any additional modules. Also checked for mechanical damage or bad soldering and didn't found anything.

PrideParrot, 2021-07-28:
[image]

While inspect and comparing board with different sources found that I have slightly different board (red circle)  also Led (D51 - Power - Yellow circle) is disabled when power on.

Found that documentation is out of date:
BIGTREETECH Octopus - PIN.pdf
BIGTREETECH Octopus - SIZE.pdf

U8 Regulator updated(?) on my board.

PrideParrot, 2021-07-31:
Seems that don't have 3.3V on U15 regulator, just ~113mV, but 5V line is Ok.
[image]

PrideParrot, 2021-08-02:
U15 regulator was replaced to same one, but still no power. Also was founded that U15 pin 5 has connected to GND somehow. After checking C110 seems it's not working properly (pin 5 via C110 connectet to GND). Capacitor don't have capacity or resistance. After unsoldering C110 I check this capacitor and it's seems Ok. But U15 pin5 still connected to GND. Seems I have connection beetwen 3.3v line ang GND.
[image]

PrideParrot, 2021-08-04:
Also communicated with BTT support. They proposed to unsolde U15 and retest for short. Without U15: 3.3V and GND - still connected, 5V and GND - not connected. This pic is just example, 3.3V and GND connection still not founded.

[image]

Valerrra, 2021-09-11:
Hi faced the same problem. "Octopus" worked the day before, the board was in standby mode, suddenly found that the stepper motors began to knock. RaspberryPi stopped seeing the board. I removed it, connected it to a laptop, but the laptop did not detect the board either. I checked for overheating, found the voltage detector heated up to 80 degrees Celsius. I checked the nearby components and found a failed diode and a 100 kΩ resistor.
[image]

callistopili, 2021-11-13:
initeresting, I got the same issue on my Octopus v1.1.

I power up the first time from the USB-C and the board was Ok.
I them programmed the new klipper wiith TMC5160-1.2 with SPI configuration(and jumpers set as 1-2(on the MSx pins).
I then Connected the driver on X axis and powre up them with the Motor Board(12V as test and one motor), but powering the electronics at the USB-C (the 5V power jumper still set as closed).

no issues, then I started the process again with a Y Z with motors, and everything was OK.
at some point but I cannot replicate anymore I added E0 drivers and motors nd weird issues appeared.
As soon I powered the board I recognized no power LEDs on at the 3.3V rail LEDs, near to the U15 regulator.
Very hot U15 and TMC driver E0 and I switched off.
I then removed the E0 driver and even if I remove all the drivers I got the board not working.

I checked with a multimeter and I noted there is a short on the VOUT of the U15 regulator, I desoldered U15 to check clearly where the short is and the pad at Vout is still shorted (resistance about 1.1Ohm) to ground.

I powered the board to check if the 5V inpunt pad is ok and the multimeter detects 5V input(pin1) but withtout U15 of course i cannot get 3.3V.
with a short ongoing  it is a no-sense to replace the U15 due to the somewhere short on the line.

any clue?

callistopili, 2021-11-13:
from the schematic, the ..3 coming out from the regulator, powers digital side of the drivers, the endstops,buffers and the CPU.
the SLEEP pin(6) for the modded/unmodded drivers doesnt seems to be the reason for this short.
the 3.3V regulator U8, seems working ok. not U15.

callistopili, 2021-11-13:
I double checked my setup, and it is likely one new tmc5160 i fitted in E0 was faulty and leaving standalone i measured the digital power pin and ground with a multimeter and sadly it was shorted draining lot of current on the 3v3. still unsure why the CPU should be affected. still dont know why but because of that, probably the CPU is fried having a short into the +3v3 power rail. I will fix  replacing a new cpu in this case.

PrideParrot, 2021-12-06:
Forgot to add. I contacted the seller to replace the board, and he agreed to the terms of payment for the delivery. I paid for the delivery, the parcel never arrived (invalid track, no response from seller).

tinkydan, 2021-12-23:
Did any one manage get the board working again? It looks like I am having the same issue after I had the power supply plugged in to a faulty power strip.

Valerrra, 2022-05-04:
yes I removed the processor and the problem disappeared on all channels

israelsu, 2022-05-06:
this is the 2nd board running in to the same issue.
all the leds on the board are on. nothing works.
No more BTT for me - i'm done with this S***

Rincewind08, 2022-10-09:
Likewise, I wasted 44$ on this board from ali express, shelved it for a couple months, went to start it and it only powers off of usb-c and will not fucking install marlin.  For god sakes BigTreeTech, I e been a loyal customer to the brand since the skr 1.3, I have a PILE of your boards, including two burnt dead gtrs, an skr 1.4, a skr 1.4 pro, a board of your driver's, and the newest thing is always a total pile of mismarked garbage.  How the fuck do you people sleep at night?

cd85233, 2022-10-20:
Hi guys just wanted to add something I found.

I had issues with no comm and board didn't show up as a port. I was able to trace it to the 3.3 regulator u15. I lifted the Vout pad and hooked it up to a power supply. I got comms back and was able to get into dfu mode. You might have something loading down your regulator or the mcu is toast.

Also, if they are using the RT-9193 and the schematic is correct, then they did not follow the recommended layout by tying BP to gnd with a bypass cap. This might be why it seems folks are having issues. The data sheet specifically states not to leave it floating.

azytzn, 2022-10-22:
> Hi guys just wanted to add something I found.
>
> I had issues with no comm and board didn't show up as a port. I was able to trace it to the 3.3 regulator u15. I lifted the Vout pad and hooked it up [...]

Hi cd85233,
I have a brand new octopus 1.1 with F446 chip. Trying to flash but pc doesn't detect it. Also cannot flash using sd card. Can u please share a photo of your card and the spot that u mod.
Thanks

cd85233, 2022-10-22:
Sure. Here it is. It's not pretty but it works. 
[image]

Sagittariin, 2022-11-01:
@cd85233 so if i desolder the U15 and do that mod  that you did , my octupus v1.1 can be saved?

ImpulziveMyztik, 2022-11-26:
Just firing up my octopus v1.1 board after having installed it in the printer. I had loaded firmware onto it perfectly fine before, but now the thing doesn't respond. I can connect to it from my pc via usb, but the serial is just an endless loop of max3421e initialization, got 255. Anybody know if this error is related to the u15 issue here? Or am I correct in understanding that the u15 issue here is actually completely disabling boards all together, like no serial console or anything?

MichalMaRozbitouDeskuNaEnderaLol, 2024-01-25:
I think its broken.