bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #54: TMC2209 UART Extruder Not Working

Seminole81, 2021-08-12:
Running this board in UART mode with TMC2209 drivers. Tried every port after 1, 2, 3 and I can't get the extruder to work for the life of me. I can get it working by plugging it into a 1, 2, 3 config, but then obviously the other motor won't work.

https://drive.google.com/drive/folders/1O26abcTria1FimlkyYNgGnCBxTUHKMMT?usp=sharing

Seminole81, 2021-08-14:
[message (2).txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/6987074/message.2.txt)
[message.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/6987075/message.txt)

Here's my code. Can't figure out why it's not working. I've plugged into motor 3 (port 4) and tried the others as well. Nothing.

Seminole81, 2021-08-16:
Solved. E0 does not work on (x2) board. Remade firmware with extruder set to E1 and works

jeffclay, 2021-08-18:
> Solved. E0 does not work on (x2) board. Remade firmware with extruder set to E1 and works

I'm new to this and looking at compiling for my first BTT board (Octopus v1.1). What do you mean by "(x2) board"? Thanks

Seminole81, 2021-08-18:
> @Seminole81
> 
> > Solved. E0 does not work on (x2) board. Remade firmware with extruder set to E1 and works
> 
> I'm new to this and looking at compiling for my first BTT board (Octopus v1.1). What do you mean by "(x2) board"? Thanks

Sorry, I mean there I had 2 board where E0 did not work. Looks to be a problem as well as there's a few more I know of on Facebook not working.

Evildude42, 2021-09-03:
Can someone post the snippet on what exactly has to be changed - i.e. how do i get Marlin to skip e0? Are you changing the pin files to point e0 to e1 pins or what?

Seminole81, 2021-09-07:
Sorry, forgot to reply to this. You just need to comment #define E0_DRIVER_TYPE and then uncomment #define E1_DRIVER_TYPE and make sure you've got a driver on the E1 port. That's all I did and it worked.

FullTiltt, 2021-11-21:
Much thanks for this post saved me some hair pulling, had the exact same issue of e0 doing nothing regardless of stepper driver installed and only having swapped boards on a previously working setup, @Seminole81 my gratitude sir.

monsterjeepn, 2022-02-13:
The root cause to this problem is an assumption BTT made within the pins_OCTOPUS_V1_common.h file. If you look through all of the stepper pins, it appears that BTT assumes most users will be running dual z motors on separate drivers (specifically, analyze the "Z2_" pin assignments versus the schematic).