bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #57: Dual Z with Dual Endstop

user-e, 2021-08-24:
Hello,
I've found no way to get this board working with dual Z driver and dual Z endstop.
The second Z driver (Z2) was recognized, but the endstop resource never be used.

My actual configuration on the board:
```
First driver slot: one TMC2209 for X motor
Second driver slot: one TMC2209 for Y motor
Third driver slot: one TMC2209 for Z1 motor
Fourth driver slot: one TMC2209 for Z2 motor
Fifth driver slot: one TMC2209 for E0 motor
```

Printer type: COREXY with two Z motor and two separated endstop

Startup and M119:
```
echo:V83 stored settings retrieved (653 bytes; crc 39905)
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
Reporting endstop status
x_min: open
y_min: open
z_min: open
```

M43 motor and endstop section:
```
PIN: PF9         M42 P89          E0_STEP_PIN                            protected 
PIN: PF10        M42 P90          E0_DIR_PIN                             protected 
PIN: PF11        M42 P91          Z_STEP_PIN                             protected 
PIN: PF12        M42 P92          X_DIR_PIN                              protected 
PIN: PF13        M42 P93          X_STEP_PIN                             protected 
PIN: PF14        M42 P94          X_ENABLE_PIN                           protected 
PIN: PF15        M42 P95          Y_ENABLE_PIN                           protected 
PIN: PG0         M42 P96          Y_STEP_PIN                             protected 
PIN: PG1         M42 P97          Y_DIR_PIN                              protected 
PIN: PG2         M42 P98          E0_ENABLE_PIN                          protected 
PIN: PG3         M42 P99          Z_DIR_PIN                              protected 
PIN: PG4         M42 P100         Z2_STEP_PIN                            protected 
PIN: PG5         M42 P101         Z_ENABLE_PIN                           protected 
PIN: PG6         M42 P102         X_MIN_PIN                              protected 
.                                 X_DIAG_PIN                             protected 
.                                 X_STOP_PIN                             protected 
PIN: PG7         M42 P103         ESP_WIFI_MODULE_RESET_PIN              Output = 1
PIN: PG8         M42 P104         ESP_WIFI_MODULE_ENABLE_PIN             Output = 1
PIN: PG9         M42 P105         Y_MIN_PIN                              protected 
.                                 Y_DIAG_PIN                             protected 
.                                 Y_STOP_PIN                             protected 
PIN: PG10        M42 P106         Z_MIN_PIN                              protected 
.                                 Z_DIAG_PIN                             protected 
.                                 Z_STOP_PIN                             protected 
PIN: PG11        M42 P107         <unused/unknown>                       Input  = 1                       
PIN: PG12        M42 P108         E0_DIAG_PIN                            Input  = 1
PIN: PG13        M42 P109         FIL_RUNOUT2_PIN                        Input  = 1
.                                 E1_DIAG_PIN                            Input  = 1
PIN: PG14        M42 P110         FIL_RUNOUT3_PIN                        Input  = 1
.                                 E2_DIAG_PIN                            Input  = 1
PIN: PG15        M42 P111         FIL_RUNOUT4_PIN                        Input  = 1
.                                 E3_DIAG_PIN                            Input  = 1

```

What i've do in the Configuration.h:

```
//===========================================================================
[... 125 lines cut ...]
  //#define INVERT_Z4_VS_Z_DIR

  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif
```

I've also tried with:
`#define Z2_USE_ENDSTOP          PG11`

But this does not work anyway.

What the printer do:
With the G28 command, the printer move down both Z motor, do the home for X and Y, and then move up just one Z motor, that can be stopped with his endstop

ba186dp, 2021-08-25:
Try looking in the section Marlin / src / pins / stm32f / pins_BTT_OCTOPUS_V1_common.hdo you have a well-defined Z2 axis: define Z2_STEP_PIN PG4 // ENGINE 3#define Z2_DIR_PIN PC1#define Z2_ENABLE_PIN PA0#ifndef Z2_CS_PIN#define Z2_CS_PIN PC7#endif Odoslané z aplikácie Pošta pre Windows Od: user-eOdoslané: utorok 24. augusta 2021 21:41Komu: bigtreetech/BIGTREETECH-OCTOPUS-V1.0Kópia: SubscribedPredmet: [bigtreetech/BIGTREETECH-OCTOPUS-V1.0] Dual Z with Dual Endstop (#57) Hello,I've found no way to get this board working with dual Z driver and dual Z endstop.The second Z driver (Z2) was recognized, but the endstop resource never be usedMy actual configuration on the board:First driver slot: one TMC2209 for X motorSecond driver slot: one TMC2209 for Y motorThird driver slot: one TMC2209 for Z1 motorFourth driver slot: one TMC2209 for Z2 motorFifth driver slot: one TMC2209 for E0 motorPrinter type: COREXY with two Z motor and two separated endstopStartup and M119:echo:V83 stored settings retrieved (653 bytes; crc 39905)Testing X connection... OKTesting Y connection... OKTesting Z connection... OKTesting Z2 connection... OKTesting E connection... OKReporting endstop statusx_min: openy_min: openz_min: openM43 motor and endstop section:PIN: PF9         M42 P89          E0_STEP_PIN                            protected PIN: PF10        M42 P90          E0_DIR_PIN                             protected PIN: PF11        M42 P91          Z_STEP_PIN                             protected PIN: PF12        M42 P92          X_DIR_PIN                              protected PIN: PF13        M42 P93          X_STEP_PIN                             protected PIN: PF14        M42 P94          X_ENABLE_PIN                           protected PIN: PF15        M42 P95          Y_ENABLE_PIN                           protected PIN: PG0         M42 P96          Y_STEP_PIN                             protected PIN: PG1         M42 P97          Y_DIR_PIN                              protected PIN: PG2         M42 P98          E0_ENABLE_PIN                          protected PIN: PG3         M42 P99          Z_DIR_PIN                              protected PIN: PG4         M42 P100         Z2_STEP_PIN                            protected PIN: PG5         M42 P101         Z_ENABLE_PIN                           protected PIN: PG6         M42 P102         X_MIN_PIN                              protected .                                 X_DIAG_PIN                             protected .                                 X_STOP_PIN                             protected PIN: PG7         M42 P103         ESP_WIFI_MODULE_RESET_PIN              Output = 1PIN: PG8         M42 P104         ESP_WIFI_MODULE_ENABLE_PIN             Output = 1PIN: PG9         M42 P105         Y_MIN_PIN                              protected .                                 Y_DIAG_PIN                             protected .                                 Y_STOP_PIN                             protected PIN: PG10        M42 P106         Z_MIN_PIN                              protected .                                 Z_DIAG_PIN                             protected .                                 Z_STOP_PIN                             protected PIN: PG11        M42 P107         <unused/unknown>                       Input  = 1                       PIN: PG12        M42 P108         E0_DIAG_PIN                            Input  = 1PIN: PG13        M42 P109         FIL_RUNOUT2_PIN                        Input  = 1.                                 E1_DIAG_PIN                            Input  = 1PIN: PG14        M42 P110         FIL_RUNOUT3_PIN                        Input  = 1.                                 E2_DIAG_PIN                            Input  = 1PIN: PG15        M42 P111         FIL_RUNOUT4_PIN                        Input  = 1.                                 E3_DIAG_PIN                            Input  = 1 What i've do in the Configuration.h://===========================================================================//============================== Endstop Settings ===========================//=========================================================================== // @section homing // Specify here all the endstop connectors that are connected to any endstop or probe.// Almost all printers will be using one per axis. Probes will use one or more of the// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.#define USE_XMIN_PLUG#define USE_YMIN_PLUG#define USE_ZMIN_PLUG//#define USE_IMIN_PLUG//#define USE_JMIN_PLUG//#define USE_KMIN_PLUG//#define USE_XMAX_PLUG//#define USE_YMAX_PLUG//#define USE_ZMAX_PLUG//#define USE_IMAX_PLUG//#define USE_JMAX_PLUG//#define USE_KMAX_PLUG // Enable pullup for all endstops to prevent a floating state#define ENDSTOPPULLUPS#if DISABLED(ENDSTOPPULLUPS)  // Disable ENDSTOPPULLUPS to set pullups individually  //#define ENDSTOPPULLUP_XMAX  //#define ENDSTOPPULLUP_YMAX  //#define ENDSTOPPULLUP_ZMAX  //#define ENDSTOPPULLUP_IMAX  //#define ENDSTOPPULLUP_JMAX  //#define ENDSTOPPULLUP_KMAX  //#define ENDSTOPPULLUP_XMIN  //#define ENDSTOPPULLUP_YMIN  //#define ENDSTOPPULLUP_ZMIN  //#define ENDSTOPPULLUP_IMIN  //#define ENDSTOPPULLUP_JMIN  //#define ENDSTOPPULLUP_KMIN  //#define ENDSTOPPULLUP_ZMIN_PROBE#endif // Enable pulldown for all endstops to prevent a floating state//#define ENDSTOPPULLDOWNS#if DISABLED(ENDSTOPPULLDOWNS)  // Disable ENDSTOPPULLDOWNS to set pulldowns individually  //#define [... cut ...]  Z3_USE_ENDSTOP        _YMAX_      #define Z3_ENDSTOP_ADJUSTMENT 0    #endif    #if NUM_Z_STEPPER_DRIVERS >= 4      #define Z4_USE_ENDSTOP        _ZMAX_      #define Z4_ENDSTOP_ADJUSTMENT 0    #endif  #endif#endifI've also tried with:#define Z2_USE_ENDSTOP PG11But this does not work anyway.What the printer do:With the G28 command, the printer move down both Z motor, do the home for X and Y, and then move up just one Z motor, that can be stopped with his endstop
	
      
		Bez virů. www.avast.com

user-e, 2021-08-25:
Thanks @ba186dp, 
your comment cleaned:

```
Try looking in the section Marlin / src / pins / stm32f / pins_BTT_OCTOPUS_V1_common.h
do you have a well-defined Z2 axis: 
#define Z2_STEP_PIN PG4 // ENGINE 3
#define Z2_DIR_PIN PC1
#define Z2_ENABLE_PIN PA0
#ifndef Z2_CS_PIN
#define Z2_CS_PIN PC7
#endif 
```

Yes, in the "Marlin/src/pins/**stm32f4**/pins_BTT_OCTOPUS_V1_common.h" i got:

```
//
// Steppers
//

#define Z2_STEP_PIN                         PG4   // MOTOR 3
#define Z2_DIR_PIN                          PC1
#define Z2_ENABLE_PIN                       PA0
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PC7
#endif

[...]

//
// Trinamic Stallguard pins
//
#define X_DIAG_PIN                          PG6   // X-STOP
#define Y_DIAG_PIN                          PG9   // Y-STOP
#define Z_DIAG_PIN                          PG10  // Z-STOP
#define Z2_DIAG_PIN                         PG11  // Z2-STOP
#define E0_DIAG_PIN                         PG12  // E0DET
#define E1_DIAG_PIN                         PG13  // E1DET
#define E2_DIAG_PIN                         PG14  // E2DET
#define E3_DIAG_PIN                         PG15  // E3DET

```

In fact the endstop are physically connected to PG11, but it never recognize it

ba186dp, 2021-08-25:
Well, I don't have physical, end stops, I'm going without them, so I have software, I'm going sensor lees. you have to play with it. And do you have the limit switch in the connector that belongs to the given stepper connected correctly? Check the connection with the PIN table again.OK let me know. Odoslané z aplikácie Pošta pre Windows Od: user-eOdoslané: streda 25. augusta 2021 9:21Komu: bigtreetech/BIGTREETECH-OCTOPUS-V1.0Kópia: ba186dp; MentionPredmet: Re: [bigtreetech/BIGTREETECH-OCTOPUS-V1.0] Dual Z with Dual Endstop (#57) Thanks @ba186dp,your comment cleaned:Try looking in the section Marlin / src / pins / stm32f / pins_BTT_OCTOPUS_V1_common.hdo you have a well-defined Z2 axis: #define Z2_STEP_PIN PG4 // ENGINE 3#define Z2_DIR_PIN PC1#define Z2_ENABLE_PIN PA0#ifndef Z2_CS_PIN#define Z2_CS_PIN PC7#endif Yes, in the "Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h" i got://// Steppers// #define Z2_STEP_PIN                         PG4   // MOTOR 3#define Z2_DIR_PIN                          PC1#define Z2_ENABLE_PIN                       PA0#ifndef Z2_CS_PIN  #define Z2_CS_PIN                         PC7#endif  [...]  //// Trinamic Stallguard pins//#define X_DIAG_PIN                          PG6   // X-STOP#define Y_DIAG_PIN                          PG9   // Y-STOP#define Z_DIAG_PIN                          PG10  // Z-STOP#define Z2_DIAG_PIN                         PG11  // Z2-STOP#define E0_DIAG_PIN                         PG12  // E0DET#define E1_DIAG_PIN                         PG13  // E1DET#define E2_DIAG_PIN                         PG14  // E2DET#define E3_DIAG_PIN                         PG15  // E3DET In fact the endstop are physically connected to PG11, but it never recognize it
	
      
		Bez virů. www.avast.com

Voodoobrew101, 2021-10-02:
uncomment INVERT_Z2_VS_Z_DIR in Configuration_adv.h

```
//
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  // Enable if Z motor direction signals are the opposite of Z1
  //#define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR

  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _ZMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif
```

Arnie3d, 2023-01-17:
i have same problems nothing seems to work z1 and z2 are parallel ( inner connected) it works without a second driver in z2 but can not change direction no z2 i have noticed that TMC2209 drivers do not work there is something wrong with this board  if i want to have z2 or y2 i define it but doesnt find the driver its a crap board only for klipper no  documents how to use marlin i,j--v.w crap doesnt work with tmc2209 someone started to mes up version 2.0.9.3 ---2.1.2 with removing num z steppers in config adv setting spend hours with this bug version drank too much coffee  blodd shot eyes maybe you can give me hope