bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #62: error

fiorettone, 2021-09-07:
hello i have a error for compile firmware on octopus with marlin and software visual studio the error is:
"TFT_CS_PIN was not declared in this scope;did you mean TFT_SPI?"

NeoMod, 2021-10-01:
sur Windows et Linux, vous ne devriez pas avoir besoin de pilotes spéciaux.
Cependant, la carte n'est détectée que comme PORT SERIE COM pour le contrôle de l'impression (par exemple avec pronterface).
La microSD n'est pas exposée correctement, je soupçonne qu'il s'agit d'un problème de bootloader. Tu vois #74 

Remarque : Postez automatiquement traduit par Google Traduction. Je m'excuse pour les erreurs.

fiorettone, 2021-10-04:
hello i have buy btt esp07s module with octopus board but dont work ! is probably to programmer? or is ready to use? if yes can you give how must configure on octopus board?