bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #64: Inductor Probe now working for either SENSOR or with Diode setup

msteele999, 2021-09-11:
I have a BTT Octopus I'm putting into my V2.4 - I have completed all my testing up to the Inductive Probe tests. I am not able to get the probe to work - it is perpetually open or closed, depending on the pin setting in printer.cfg - I have replace the probe and am having the exact same issue.

First attempts were with a PL-05N2 NPN NC probe. I tried wiring it direct to the SENSOR Port (Black=SIG, Blue=GND, Brown=+V). I then added a pull-up 4.7K ohm resistor across SIG and +V. The LED would illuminate but the probe would not trigger. I then wired it as +V and GND on SENSOR, and SIG on pin PG15, DIAG7 with BAT85 diode in circuit, band facing the PROBE - the LED would not illuminate at all and the probe would not trigger.

I then got a PL-05N NPN NO probe.  I repeated all the steps above, however this time, even when wired to SENSOR, the LED never illuminated and all my tests were negative.

I have measured the voltage across +V and GND at the board and it is +23.3 VDC. I have measured the voltage across +V and GND at the probe connections and it is also +23.3 VDC.

I don't know where to go from here - is there a different type of probe I should try or some other wiring I need to use? Is it possible the SENSOR port is bad somehow?

Oh - just a note - when running with the diode, I changed the pin: PG15, when running SIG on the Sensor, I changed the pin: ~PB7 in printer.cfg

To recap: no LED light on probe now (I did at first, now nothing), 23.3 VDC across the SENSOR port on Octopus 1.1, tested both diode to PG15 and a 4.7K resistor across the octocoupler - no joy.

msteele999, 2021-09-11:
One other note (not sure if it means anything):

- when I measure the voltage of +V and GND with the probe out of circuit, I measure 23.3VDC. When I put the probe in circuit, voltage across +V and GND drops to 1.1VDC - regardless if there is metal in front of the probe or not.

msteele999, 2021-09-11:
Additional information:

When I connect the probe +V and GND directly to the Meanwell PSU, the LED illuminates and the probe triggers in proximity to metal.

When plugged into the SENSOR pins it does not.

I have measured the voltage across +V and GND on the SENSOR pins and it is +23.3VDC

It's acting like there is some circuitry at the SENSOR pins that is not working

msteele999, 2021-09-11:
OK - my tests:

- Probe wired directly to +V and GND on PSU
- SIG wired to pin PG15 of DIAG7 block , J34
- printer.cfg pin changed to ^PG15

Probe works!

[image]

Using SENSOR port in any configuration results in NO LED, NO probe state change.

Please advise.

msteele999, 2021-09-11:
One last note (hopefully) - I have been able to successfully run it on a fan port as well

3Dgraf64, 2021-10-29:
If your probe is connected to J40 you have to select the proper voltage with the jumper J38.

rmilyard, 2021-11-26:
I am having this issue.  I have my probe J38 jumper for 24v.  I have sensor on PB7 pin but always shows OPEN.

thaers3d, 2021-12-12:
i have the same problem ,
when i disconnect the signal wire from the board the sensor work normal but when i connect again it will not work.
please any body have solution for this problem?

deece, 2021-12-15:
I also have the same problem, V+ drops to 5V when the probe is connected, 24V when it's not. Moving the 24V line to one of the fan headers fixes the problem.

jpyflight, 2021-12-19:
Hi all, was experiencing same issues with PL-05N. I am using the Hartk 4.0 PCB, so V+ and ground are routed somewhere else already. Diode is soldered on at the tool head. Signal pin was connected to J40, pin PB7; no results. Lights on, metal close to the sensor triggers it physically, but nothing on mainsail.  2nd try, signal pin was connected to J33, pin PG11; again same results, physically works but nothing on mainsail. 3rd and successful configuration, signal pin connected to J34, pin PG15. Light stays on when open, shuts off when triggered. Works and shows in mainsail. 

## For the spare endstop port; use this with the Fotek PL-05N (or generic) inductive probe
pin: ^PG15

[image]

martynov-ss, 2022-09-14:
Simply connect the sensor to the J40 connector normally, and move the signal wire from the sensor to the PB7 of the J43 connector (bltouch). It works.

tweakstar, 2022-10-18:
> Simply connect the sensor to the J40 connector normally, and move the signal wire from the sensor to the PB7 of the J43 connector (bltouch). It works.

do you need a pull up resistor for this connection method?

erycstrife, 2023-01-10:
Hi did that work?