bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #70: MKS_MINI_12864 not configured in pins_BTT_OCTOPUS_V1_common.h

Gleb1982, 2021-09-25:
This leads to compiling error.

NeoMod, 2021-09-29:
+1 this issue renders unusable the  12864A/B LCD

Gleb1982, 2021-09-29:
I solved the problem by adding 

```
 #elif ENABLED(MKS_MINI_12864)

    #define DOGLCD_A0                EXP1_04_PIN
    #define DOGLCD_CS                EXP1_05_PIN
    #define BTN_EN1                  EXP2_08_PIN
    #define BTN_EN2                  EXP2_06_PIN
```

And rotating EXP1 and EXP2 180 degree (I just cut the key on one side of both wires).

DrakulaRS, 2021-09-30:
Где это прописать? подскажите!

NeoMod, 2021-10-01:
@Gleb1982 thanks for the quick tip. It's working, but I hate having to meddle with PINS definition just because devs are not paying attention. I guess this will be another "shelf purchase".

@DrakulaRS locate the file `pins_BTT_OCTOPUS_V1_common-h` in `Marlin/scr/pins/stm32f4` ad and proceed adding the section mentioned early.  Look around line 444 for the right spot to insert those new lines.

Also, reading the PINS is useful to learn about the "peculiar" - to say the least - way the assigned names and positions: it will come in handy when compiling your own firmware since you can't freely define assignments based on "PIN ID" inside configuration.

NeoMod, 2021-10-01:
I've made a pull request with the quick fix integrated. #75