bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #73: BLtouch not working

To0wnn, 2021-09-29:
Hi,

Trying to get my Ender 5 Plus to work with the octopus V1.1 with marlin_bugfix
But somehow the BLtouch is not working.
When startup the pin comes out and retracts again.
But when I try to manually deploy the pin through the screen it is not working. (When homing it is also not working).
Wiring is correct and also tried with a different BLtouch also not working, looked also for the wiring in the manual.
BLtouch is working with a different board(SKR PRO V1.1). 
Also tried to connect to the Z_MIN_ENDSTOP_PIN (PG10) not working (with the changes is the marlin code).
Any suggestions?
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/7255276/Configuration.zip)

ba186dp, 2021-09-30:
Hi,
I have the same problem with BLTOUCH, if I give it firmware that is set to basic data only, BLTOUCH goes as I suddenly compile what I need for my machine (dual X, dual Z, dual extruder, stealt shop and other necessary things), BL TOUCH doesn't work for the world either, I've already tried almost everything I could think of. The connection to the Z_MIN slot (firmware change) does not work, yes in the original BLTOUCH slot from the manufacturer. I am clueless and frustrated with what is happening with this board or firmware. Please advise.

iworld4us, 2021-09-30:
Try my version:
https://github.com/iworld4us/RepRapFirmware-OCTOPUS-V1.1

ba186dp, 2021-09-30:
Thanks, but the RepRAP firmware is a big unknown to me, but I'll probably have to start getting to know each other, because MARLIN is one big mess. THANK YOU

NeoMod, 2021-10-01:
That's strange: on my v1.1 the BLTouch is working as expected. Have you tried enabling the `#define EDITABLE_SERVO_ANGLES` ?

ba186dp, 2021-10-01:
YES I tried, it doesn't work, after the reset it flashes blue, when trying to level it it only lights up red. I don't know what's wrong with that. If it works for you, what you set in Marlin, and where you have it involved, I'm in the original slot on BL TOUCH, no Z_MIN.

NeoMod, 2021-10-01:
I'm connected to the same port _(the dedicated one, let's call by name J43  so anyone reading can't be confused)_ the only difference I could think of, with your config, is that I don't have a TFT connected to the board, so the adjacent port J70 is empty.

Nothing peculiar in my config either, just the classic:

`#define BLTOUCH`

`#define BLTOUCH_SET_5V_MODE`

I'm not specifying a `BLTOUCH_DELAY` since the board treats the J43 port as the sole `SERVO_PORT` so the delay should be applied from the `#define SERVO_DELAY` line in _Configuration.h_.

_(In the official documentation the servo delay is set to 300, you might want to try with this value removing the dedicated BLTOUCH_DELAY.)_

One thing that gives me some thoughts is the behaviour you described: in my experience, that pattern of LED behaviour indicates a missing and/or irregular signal/ground connection.

I can imagine you've already tried it, but did you check that little JST-like connector on the back of the BLTouch? It has a very brittle retention system and I've seen it - on some units - getting loose on one end, leaving the connector a few millimetres out of the socket.

Also - that's the OCD in me speaking - have you considered re-making the terminal plug with a proper JST plug? I have terrible experiences when it comes to dupoint connectors, especially those of 3rd party origin.

Although since the same BLTouch is working with another board, these two suggestions might well be automatically excluded.

On the other hand, have you tried providing the board with an external power supply? (12V or 24V) Just as a "last chance" scenario.

ba186dp, 2021-10-01:
The power supply is 24 V 500 w, I have another BLTOUCH, which is not firmly connected to the printer and the connecting wires do not go through the JST connector, but are firmly soldered to BL TOUCH, I will try BLTOUCH 5V. I'll see I'll try.
Marlin latest clean version, compiled once.
I can attach both config files, and if I can ask, check them out for me, because I'm already crazy about it. Thank you in advance
[Configuration_advh.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/7267833/Configuration_advh.txt)
[Configurationh.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/7267834/Configurationh.txt)
.

To0wnn, 2021-10-01:
> I'm connected to the same port _(the dedicated one, let's call by name J43 so anyone reading can't be confused)_ the only difference I could think of, with your config, is that I don't have a TFT [...]

Nope not working for me as wel.
Increased `BLTOUCH_DELAY` to 600 also tried with `SERVO_DELAY` to 600 still not responding
But is doing something additional during startup
The LED is steady red not blinking

> Try my version: https://github.com/iworld4us/RepRapFirmware-OCTOPUS-V1.1

Did take the compiled version from BTT
BLTouch also working and responding to the commands awsome!!!
But my TMC5160 are not working :(
Any suggestions?

[board.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/7267764/board.txt)
[config.g.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/7267772/config.g.txt)

NeoMod, 2021-10-01:
> The power supply is 24 V 500 w, I have another BLTOUCH, which is not firmly connected to the printer and the connecting wires do not go through the JST connector, but are firmly soldered to BL [...]

I have a bit of time, I'll try and compare the files.

At quick glance, I noticed you have enabled:

`#define ENDSTOPPULLUP_ZMIN_PROBE`

I'm not using this feature with my BLTouch. Are sure about it?

ba186dp, 2021-10-01:
I'm not sure about anything anymore, I'm connecting the output from M119, the voltage is at pin 4.997 V. I'll try the Probe pullu
[image]
p.

NeoMod, 2021-10-01:
BTW, it seems that there is something more customized than PINS and Config in the Marlin release from BTT on this GIT _(*)_. The latest official release from Marlin won't run properly on the board and drives the BLTouch crazy.

Probably I'm missing something but I can't seep through all the files to identify the changes, so for now I would advice compiling and testing with the provided Marlin folder (_Firmware/Marlin-bugfix-2.0.x/_)

***=** (If that's the case, I really don't like it. BTT should PULL their changes into Marlin official)

To0wnn, 2021-10-01:
> BTW, it seems that there is something more customized than PINS and Config in the Marlin release from BTT on this GIT _(*)_. The latest official release from Marlin won't run properly on the board [...]

Nope same issue.. 
Can you maybe give your firmware.bin?
I get the idea that my vscode is doing something strange.

NeoMod, 2021-10-01:
> Nope same issue.. Can you maybe give your firmware.bin? I get the idea that my vscode is doing something strange.

Sure, here it is: [firmware.bin](https://drive.google.com/file/d/10WhOnH8JkmQ0I1lGrY4yScdeZAm9T_-i/view?usp=sharing)

It's compiled for my printer, so without LCD/TFT support (I'm still waiting on that delivery) and it uses some spare ports (J30/J32/J34) for a single RGB Led with PRINTER_EVENT_LEDS active. But for just testing the BLTouch it should be fine.

Also, I compiled yours using the two files you shared early: [firmwareTEST.bin](https://drive.google.com/file/d/1v3-Pem6rfWO9A_M733V7qTd-6ZByBtOP/view?usp=sharing) Please note that _**I haven't checked anything, I just compiled based on your Configuration and Configuration_adv files.**_

ba186dp, 2021-10-01:
Guys, you're very golden that you showed a big heart and helped me, everything works as it should, the problem is the poor quality of the BLTOUCH product, the cold connections on the PCB, I normally threw it in the recycling bin and used it from the second BLTOUCH printer, which goes 100%. Thank you and I appreciate it. There are social and professional networks for this, HELP if you can't advise.

To0wnn, 2021-10-01:
Thanks for al the help!! It was firmware related. Thank you for the bin file it really helped me! I switched in the pin config the BED and FAN2 pin because i have an external mosfet. When I comment out FAN2 en give the BED the pin of  FAN2 everything is working!! 

Also my suspicion but not sure if it was the issue was but I deleted the folders ".pio" and ".vscode" from the folder where Marlin is standing. Normally I always pressed clean in VScode

closed 2021-10-01 by To0wnn

NorbertRoller, 2022-03-02:
With the BTT Firmware (BIGTREETECH-OCTOPUS-V1.0-master.zip) and the changes recommended above,  I had to connect the black & white BLTouch cable to the z-endstop instead of the J43 to make it work.