bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #74: Brand new Octopus v1.1 - microSD detection over USB not working

NeoMod, 2021-10-01:
I'm reporting a pretty annoying bug, not seeing it explicitly mentioned in any other issue. 

On a brand new motherboard, the microSD card is not made available via USB (USB-C) to the host thus forcing the user to physically take out the card every time a firmware update is needed.

Things I've tried with no success:

1. changing microSD card
2. changing USB-C cable, including the one provided in the package
3. changing USB port
4. changing host (two windows hosts shows the same problem)
5. changing host and OS: tested on Raspberry Pi 4 with Debian, microSD is not available.
6. testing with dfu-utils on Raspberry: same result, board is not available.
7. checking all the required drivers are installed: motherboard shows up correctly as a "COM PORT" and can be controlled via Pronterface. But microSD card is not available.
8. updating bootloader, as per instructions provided here

Seems definitely an issue with the motherboard, probably with the bootloader (?) since I even tried two different models of the same: one coming from Aliexpress and one from Amazon. They both behave in the very same way.

Just a few more clarifications:

- board is "vanilla" out of the box: no component added, no jumper moved, no firmware loaded
- on a second board I tried loading the provided Marlin firmware from GitHub to see if it would make a difference: no difference at all in behaviour.
-  tried with both USB Power and External Power (24V)
- tried with and without "power jumpercap" when using external power
- as far as I can tell, all other aspects of the board are working as intended. no visible components damaged nor soldering issues visible on both the boards tested.
- the microSD cards used have been formatted multiple times according to standard and they have been checked for consistency. 

It's quite an annoying bug for a 50€+ board, especially considering low-end boards (8bit MCU even) are perfectly capable of this feature.
I do want to believe this is a bug and not an "intended feature"...if so I missed the explicit warning/disclosure about it.

iworld4us, 2021-10-01:
Try fat16 or fat32 or format it in Windows

NeoMod, 2021-10-01:
Thanks, but I've already made sure of it. I even used SDCardFormatter.