bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #77: Unable to read tmc uart 'stepper_z1' register IFCNT

Dedelkov, 2021-10-11:
I keep getting this error on Klipper, and can't figure out why. 

I'm using the default .cfg file for the Voron 2.4

It only happens on the 4th driver position (the default for the Z1 motor) on the Octopus Board.

I have another six TMC2209 stepper drivers configured with uart working without issues in the same board.

matticustard, 2021-10-13:
Is it possible you skipped a driver slot and put one in the top slot instead? This can be a little confusing since you are supposed to skip a motor connection, but not a driver connection. The drivers should all be positioned side-by-side as they are in this picture.

[image]

chefguevara, 2021-10-31:
I have the same issue currently, same board, it started all of the sudden, it was printing just fine. Drivers are in the right positions and wires too skipping the z1-motor 2 output as intended. 

The only difference with the previous time it worked is that I wired the controller fan and LCD. If it works for something when the error pops up when I send the printer to home any axis, the LCD screen goes blank and all fans goes to 100%.

My next steps will be disconnect the fan and screen and try again, I'll update if I see any change.

dspublic, 2021-12-13:
Hi,
Same proble here. When I apply [display] parameters in klipper , then I get "Unable to read tmc uart 'stepper_xxx' register IFCNT" error. But, the 12864 mini display isn't attached, only configured. So, I think it's probably a klipper issue...

@chefguevara Did you solve the problem?

dspublic, 2021-12-13:
@chefguevara Finally, I have found the mistake. I left TCM jumpers in SPI mode, instead of UART!

chefguevara, 2022-01-13:
Solve!

Ok. At first I was like "naaah I have a bag full of jumpers to remind me that I have all the drivers in the right position, that's not my case". 

But it seems that you're absolutely correct @dspublic! I missed one jumper. Is hard to see the current setup with all the drivers plugged in so I only noticed the mistake after removing all of them to triple check. Interesting enough, the printer works just fine with only one driver misconfigured with all the jumpers in place. It just interferes with the screen apparently.

TommySchwartz, 2022-10-26:
I have this " Unable to read tmc uart 'stepper_a' register IFCNT" only after homing to quickly after aborting a print. Not always stepper_a, but it is the most common.

Have a delta printer, so I have stepper_a, stepper_b and stepper_c.

Otherwise the printer works great.

torwag, 2023-06-08:
Just to leave a message here for people as stupid as me:
The error message also pops up, if you haven't powered up the printers main supply, but only connected via USB for example with klipper and mainsail. Being able to "remote control" the printer, it might not be all that obvious that the printer runs out of power, trying to home.