bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #78: Analog Input

jasonhamilton, 2021-10-15:
I’d like to connect my E3D PT100 amplifier board to an analog input.  The documentation indicates:

Temperature sensor interfaces: TB, T0, T1, T2, T3 (With direct inputs available that bypass the protection circuit for alternate use).

How can I access the direct input? Does this disable to voltage divider?

Thanks!

c3charvat, 2021-10-31:
Pin header J47 (see schematics) it's just below the others. Is labeled pt100 and is connected to that circuit though there is no reference if it's a direct connection or not. (But by contrast there is for every other one of the sensor interface you mentioned)

jasonhamilton, 2021-11-09:
@c3charvat the pin J47 is connected to the amplifier circuit where you can solder an surface mount INA826 AIDR amplifier and not accessible via any of the headers.  I was able to get this to work but I had to tap into one of the solder pads connected directly to the controller's J47 pin.

I'll leave some additional information in case anyone else may need it.  Here is the pad that's connected to the ADC pin J47.

### J47 Pin

[image]
This is a 3.3V ADC, so make sure you set this in the firmware accordingly.

### Using a Thermistor Port

The other option is to use a thermistor pin but this requires some soldering work.  I've mapped out the thermistor circuit below. 

[image]

[image]

Each thermistor has its own circuit.  Because this is a voltage divider circuit for use with thermistors you will need to remove it to get a proper reading.  In order to do this you would need to remove the 5 components below and then put a junction between the TH and THERM pins.

closed 2021-11-09 by jasonhamilton

c3charvat, 2021-11-09:
Glad I was able to help some what!