bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #83: Can't compile BIGTREE_OCTOPUS_V1_USB

hololight, 2021-11-04:
I can compile BIGTREEE_OCTOPUS_V1 but not the ...USB version. I believe the first error is:

C:\Users\holol\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_desc.c:146:23: error: redefinition of 'USBD_Class_DeviceDesc'

But prior to that, I get a ton of Warnings, which I dont get with the non-USB build. Since I thought it might be something I had done, I downloaded a fresh copy from the repo and attempted to compile... same situation... non-usb compiles fine, usb has ton of warnings, and eventually errors out.

Am I doing something wrong?

the USB version IS required to be able to connect up my Pi and use octoprint, correct?

exxosuk, 2021-11-05:
I can't get it to compile either, but the other one does (but doesn't actually work in the board) 

` V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x> Environment             Status    Duration
The term 'Environment' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the n
ame, or if a path was included, verify that the path is correct and try again.
At line:1 char:12
+ Environment <<<<              Status    Duration
    + CategoryInfo          : ObjectNotFound: (Environment:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x> ----------------------  --------  ------------
>> BIGTREE_OCTOPUS_V1_USB  FAILED    00:00:13.993
>> ================================================= 1 failed, 0 succeeded in 00:00:13.993 =================================================
>> PS V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x>`

hololight, 2021-11-05:
Well I don't know what the deal is now.... Since I had no better options, I loaded up the first firmware build and my board seems to be working (at least its responding properly in pronterface over USB (com port)). But that tells

exxosuk, 2021-11-05:
Same here, no idea what the USB one is for. There never seems to be any proper guides explaining what all this stuff is for.  I'm still trying to work out why the other firmware compiles but doesn't work.

hololight, 2021-11-05:
I'm thinking the ...USB build target is perhaps related to the onboard USB-A port? Maybe having to do with either allowing Flash Drives, or perhaps USB-OTG... regardless, if the first build target allows the board to communicate over USB (as evidenced by Pronterface), then I would think I'm good to go with Octoprint working now. What do you mean by 'doesn't work'? I'm just wondering if despite the Pronterface responses from the board, it wont actually work once I put it in my printer and hook everything up. Do you want me to give you my config file?

exxosuk, 2021-11-05:
I opened up a new topic for my issue  https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/85

I am using the files as in the repo. I would have thought it would all be setup to compile to the firmware.bin as also included. But my file ends up at 159K not 480K.

hololight, 2021-11-05:
Mine is 211K, and there were a lot of changes I had to make

hololight, 2021-11-05:
@exxosuk The firmware that got you running was mine? Do note, that my bed is 400x400mm, so thats what's coded in the firmware.

exxosuk, 2021-11-05:
It was mine. I've not set the bed area up yet. Just got homing XYZ to work which is  a big step in the right direction!  

I have no idea what the USB file is for. Maybe it's not supposed to be there.. ?

hololight, 2021-11-05:
Which config file? A fresh one you made, or one based off mine? (basically i'm looking for how close I will be when I try my board out)

exxosuk, 2021-11-05:
It's the one in the default repo files. It does compile and work without changes afterall. 

The only changes I have made so far are inverting the end stop switches, and changing the homing direction. But this is specific for my printer anyway.