bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #85: Compiling works but board does not function

exxosuk, 2021-11-05:
The BG file size is 480K, but when I compile, it comes out as 159K ?? If I flash to the board and send M115..

`SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Nov  4 2021 22:39:38) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:4 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff`

It has loaded the new file. But the printer is otherwise "dead". Trying to move the motors does nothing at all. 

I am using VScode on windows.. is there some trick to getting it to compile or something ? Or should I be using the USB version (which doesn't compile at all) .

hololight, 2021-11-05:
The steppers aren't setup by default, you've made those changes in the config?

exxosuk, 2021-11-05:
I've not made any changes. I've no idea what I am supposed to be doing. I wish creators would just upload a "ready to compile" set of files, so we can just make the changes we need without having to mess about for endless hours trying to figure out why nothing works :(

exxosuk, 2021-11-05:
ahh so assume this is part of my problem ?

[image]

hololight, 2021-11-05:
good catch (I didn't have that line set like I should have), lol. I've forked the repo from BTT and in the firmware folder I've uploaded my configuration.h and configuration_adv.h. Do you know how to compare two files in VSCode? (that way you can compare my files with the defaults to see the changes I've made)

Can you see if you can talk to the board over USB (like with pronterface)?

exxosuk, 2021-11-05:
It will connect and respond to command M115 as stated in my first post. The time stamp shows it has been updated with my compiled firmware. But other than that no movement commands work. 

It does however work fine if I use the firmware.bin (480K file size)  . So you would think the files in the repo would compile directly to that file size and all be set up for the octopus board.

I don't know how to file compare, but I know how to do it in winhex.

hololight, 2021-11-05:
In VSCode press <F1> to bring up the 'command window', then start typing "Compare active file with" and it'll give you like 3 options, the first should be the one you want. You need to have had both versions open in the same Code window.

Where did you find the precompiled version of the firmware? I didn't notice it in the repo.

exxosuk, 2021-11-05:
The second one down I used. 

[image]

exxosuk, 2021-11-05:
Now putting the 480K one in doesn't work anymore :(

`Connecting...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
Unable to read board configuration: 0:/sys/board.txt...
`

hololight, 2021-11-05:
Maybe I was confused, but the firmware you're trying to load (that worked before... that firmware.bin) is RepRap firmware, not Marlin... that's why its such a different size

exxosuk, 2021-11-05:
I've lost the plot. I even put the 480K one on the SD card folder as "OK" folder as I was sure it worked. So no idea what I actually had in the board now. I've tried both the 480K ones and neither work. 

Maybe the only one which would work is the firmware-F446-USB.bin one, that's 20K.. this is just madness.

exxosuk, 2021-11-05:
That firmware does even less. Windows won't even install the driver now. 

I don't suppose you have the original firmware.bin which was shipped with the board ?

hololight, 2021-11-05:
The firmware file that is in "...Octopus works on Voron v2.4\Firmware\RRF" is actually RepRapFirmware not Marlin. Its just another firmware alternative, just like Klipper.... but I dont know anything about either of those two... and I'm still at the beginning of my 'journey' with Marlin... I see no precompiled version of Marlin based firmware in the repo... And so far as I know, the board shipped blank (I never tried powering it till I put in my SD card with my compiled firmware image). I can put my firmware file up on my repo, just for you to be able to test things. I was just playing with it (still not in my machine) and its working with a BTT TFT50 touchscreen connected. It is turning a fan on and off, as well as reading from a thermistor...

exxosuk, 2021-11-05:
The board was shipped with some firmware. I hooked up the stepper motors and then stop switches which was all working. But I needed to invert the switch in the firmware as open and closed is backwards. Hence the total dilemma. Because I am converting a flash forge printer, and trying to reuse as much of the existing stuff as possible. So physically changing the switch not so simple without changing even more boards and cables. As I've been at this for almost 3 months now I thought it would just be easier to do it in firmware.

I was previously using a smoothie board where all the settings were configurable in a single file which was great. Unfortunately that board had some bizarre faults where ultimately I gave up with it and brought the BTT one. This one worked great out of the box. Only now my board is basically bricked :(

It would be appreciated if you have a firmware which could get me back up and running again least. I still need to wire up the heated bed and yet so I cannot even do that currently while puzzling over the firmware compiling problems.

hololight, 2021-11-05:
I dont know that this will work for you in any way.... I dont even know that this will move my steppers once I hook it up. I just know it at least paritally works... 
https://github.com/hololight/Pre-compiled-Marlin-firmware-for-my-Octopus-v1.1-Mainboard

exxosuk, 2021-11-05:
Thanks, I tried your firmware file, and it does exactly the same as mine does :(

hololight, 2021-11-05:
Just isn't running the steppers, or no longer any response at all?

exxosuk, 2021-11-05:
I think it was just not responding to the steppers. I am trying to compile again but VScode just seems to be stuck in some loop now. 

I have tried to use the Arduino compiler but that does does not work either and complains about stuff I have no idea how to fix either.

exxosuk, 2021-11-05:
So with my 159K file..

Connecting...
Printer is now online.
Using tool 0.
echo:Active Extruder: 0

>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Nov  5 2021 15:40:43) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:4 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0

So the stepper motors are not working. But it is still odd that the LED on the board is flashing when it was not originally.

exxosuk, 2021-11-05:
This is what I get when trying to compile the USB version.

Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. All rights reserved.

PS V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x> platformio run -e BIGTREE_OCTOPUS_V1_USB
Processing BIGTREE_OCTOPUS_V1_USB (platform: ststm32@~12.1; board: marlin_BigTree_Octopus_v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_BigTree_Octopus_v1.html
PLATFORM: ST STM32 (12.1.1) > STM32F446ZE (128k RAM. 512k Flash)
HARDWARE: STM32F446ZET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 0.0.0
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.5
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb\usbd_desc.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb\usbd_ep_conf.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb\usbd_if.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb_host\hid\usb_host_hid.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb_host\hid\usb_host_hid_keybd.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb_host\hid\usb_host_hid_mouse.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb_host\hid\usb_host_hid_parser.c.o
Compiling .pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb_host\msc\usb_host_msc.c.o
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_desc.c:146:23: error: redefinition of 'USBD_Class_DeviceDesc'
  146 | __ALIGN_BEGIN uint8_t USBD_Class_DeviceDesc[USB_LEN_DEV_DESC] __ALIGN_END = { \
      |                       ^~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_desc.c:176:3: note: in expansion of macro 'USBD_CLASS_DEVICE_DESCRIPTOR'
  176 |   USBD_CLASS_DEVICE_DESCRIPTOR(USB_CDC_CLASS_MULTI, CDC_SUBCLASS_ACM, CDC_PROTOCOL_V25TER);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_desc.c:146:23: note: previous definition of 'USBD_Class_DeviceDesc' was here
  146 | __ALIGN_BEGIN uint8_t USBD_Class_DeviceDesc[USB_LEN_DEV_DESC] __ALIGN_END = { \
      |                       ^~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_desc.c:172:3: note: in expansion of macro 'USBD_CLASS_DEVICE_DESCRIPTOR'
  172 |   USBD_CLASS_DEVICE_DESCRIPTOR(0x02, 0x02, 0x00);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\BIGTREE_OCTOPUS_V1_USB\FrameworkArduino\stm32\usb\usbd_desc.c.o] Error 1
In file included from C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.c:20:
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:72: warning: "CDC_IN_EP" redefined
   72 |   #define CDC_IN_EP                     0x81U /*  EP1 for CDC data IN */
      | 
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:43: note: this is the location of the previous definition
   43 |   #define CDC_IN_EP                     0x82U  /* EP1 for data IN */
      | 
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:73: warning: "CDC_CMD_EP" redefined
   73 |   #define CDC_CMD_EP                    0x82U /*  EP2 for CDC commands */
      | 
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:44: note: this is the location of the previous definition
   44 |   #define CDC_CMD_EP                    0x83U  /* EP2 for CDC commands */
      | 
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:78: warning: "DEV_NUM_EP" redefined
   78 |   #define DEV_NUM_EP                    0x06U   /* Device Endpoints number including EP0 */
      | 
C:\Users\chris\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32\usb\usbd_ep_conf.h:46: note: this is the location of the previous definition
   46 |   #define DEV_NUM_EP                    0x04U   /* Device Endpoints number including EP0 */
      | 
[... cut ...]
================================================================ [FAILED] Took 9.69 seconds ================================================================

Environment             Status    Duration
----------------------  --------  ------------
BIGTREE_OCTOPUS_V1_USB  FAILED    00:00:09.690
========================================================== 1 failed, 0 succeeded in 00:00:09.690 ==========================================================
PS V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x> echo "done" >"C:\Users\chris\AppData\Local\Temp\ipc"
PS V:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x>

hololight, 2021-11-05:
So it could be a hardware problem, or it could be my config (a real possibility... this is only the second board I've tried to compile Marlin for).

Yeah, you're trying to compile the 'BIGTREE_OCTOPUS_V1_USB' build target?.... I've never managed to get that one to build.

exxosuk, 2021-11-05:
The partly working one compiles as shown above, the USB one fails to compile at all.

It is not a hardware problem because as I said before, I had the switches and motors all running when I use the board out of the box. The problem started when I tried to change the firmware to invert the end stop switches. Now the motors do not function at all.

hololight, 2021-11-05:
By hardware problem, I more meant since you started working with it. Static discharge, etc... But its more likely that its my config. BTW, my light blinks as well, I wasn't terribly worried about it though...

Once I get home and get the board hooked up to my printer (I only have the board, LCD and thermistors/fans with me), I'll test, then if I'm getting same result as you, I'll try and figure out what could be missing in the config...

exxosuk, 2021-11-05:
So this is some progress now, Z and Y actually work but X does not, and X was what I was clicking on to test :-\  So not sure what is going on there, but obviously this firmware does actually work it seems..

closed 2021-11-05 by exxosuk

Hecke-r, 2023-05-14:
I know I am 2 years or so late but my problem was that I hadn't connected the jumper pins to enable the uart