bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #89: BTT Octopus V1.1 Running TMC2209 in UART mode and cannot get my dual Z steppers and E0 to run independently

jmarkscnc, 2021-11-28:
I have been working on this issue for way to long, but I have 2- Z motors and the E0 all move at the same time? I have check everything I can think of, config.h, config_adv_h, along with the pins file along with watching about 40 hours of videos that have nothing to do with this board with no success. I have uploaded my Marlin build to my shared folder on One Drive if anyone can look it over or make some suggestion, I would be very much appreciated.

https://1drv.ms/u/s!An3HLl2sqsgkiGBMwvWDTJpdLMDs?e=rRv7Om

radek8, 2021-11-28:
How are you physically involved?
X-axis - Engine 0
Y-axis - Motor 1
Z-axis - Engine 2
Z2 axis - Engine 3
E0 axis - Motor 4

If so, please try.
1. if you move the Z axis, both Z motors and also E0 will move?
2. if you move the E0 axis, do both Z motors or only one move?

jmarkscnc, 2021-11-28:
My apologies for not quit understanding your first question. I am configuring this board in Marlin.

X-axis - Engine 0 --------- X-axis - Motor 0
Y-axis - Motor 1 --------- Y-axis - Motor 1
Z-axis - Engine 2 --------- Z-axis - Motor 2
Z2 axis - Engine 3 --------- Z-axis - Motor 3
E0 axis - Motor 4 --------- E0-axis - Motor 4

I am not sure were to change this if that is what you are saying. Is this something you what me to change? if so were?

Q: if you move the Z axis, both Z motors and also E0 will move? 

Yes, all of the motors turn at the same rate. I will try and attach a video link if that would help.
https://1drv.ms/u/s!Ale6SI7zYAvsbx51jUrkgCDS2bM?e=eVzpEy 

Thank you for taking the time to help on this matter.

radek8, 2021-11-28:
If you eject the filament (activate motor E0) the motors Z will move? One, both mebo none?

jmarkscnc, 2021-11-28:
When selecting unload, All of the motors move Z, Z2, and E0.

radek8, 2021-11-28:
I did not notice any errors in the configuration that would cause this problem.
You can take a good look at the board, if there is no short circuit on it?
I don't know, but I'd say it may be a HW board problem.

radek8, 2021-11-28:
If position motor 4 were shorted to the Z axis, you could remap the extruder to position motor5 in the pin set and use this position for the extruder....

jmarkscnc, 2021-11-28:
The Z and Z2 will not run independently of each other either, This is a new board and I have not been able to get this to work even when I was bench testing. 

The Pin file looks good according to the documentation?

// Steppers
//
#define X_STEP_PIN                          PF13  // MOTOR 0
#define X_DIR_PIN                           PF12
#define X_ENABLE_PIN                        PF14
#ifndef X_CS_PIN
  #define X_CS_PIN                          PC4
#endif

#define Y_STEP_PIN                          PG0   // MOTOR 1
#define Y_DIR_PIN                           PG1
#define Y_ENABLE_PIN                        PF15
#ifndef Y_CS_PIN
  #define Y_CS_PIN                          PD11
#endif

#define Z_STEP_PIN                          PF11  // MOTOR 2
#define Z_DIR_PIN                           PG3
#define Z_ENABLE_PIN                        PG5
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PC6
#endif

#define Z2_STEP_PIN                         PG4   // MOTOR 3
#define Z2_DIR_PIN                          PC1
#define Z2_ENABLE_PIN                       PA0
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PC7
#endif

#define E0_STEP_PIN                         PF9   // MOTOR 4
#define E0_DIR_PIN                          PF10
#define E0_ENABLE_PIN                       PG2
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PF2
#endif

    Is there anything else I can try before returning this Octopus board. I really am starting with an old just frame and bulding and adding for testing for a new design and I needed the functionality of this Octopus board? Just frustrated I usually can figure these types of things out but I am stumped.
    If the Motor 4 is shorted and I move it to Motor 5 that still is not fixing the issue with motors 2 and 3. They will not run independently for Z leveling either?

jmarkscnc, 2021-11-29:
Well I just bought 2 more BTT-Octopus V1.1 boards. I still think that there is something wrong with this one I got. I have never had this many issues with any BigTreeTech products so I am throwing in the towel as we say on this one. I do not see anything as far as solder bridging or anything burnt or looks shorted out. I did move and remapped the pins and I will give that a try but I'm not sure if the driver is bad and shorting something on the main board  or it's the main octopus board? Thank you for all of you assistance. I will get the new ones in a couple of days so I will reach back if this continues.

radek8, 2021-11-29:
See #54

radek8, 2021-11-29:
Send the photo plugged in. 
Have you used the right connectors?

jmarkscnc, 2021-12-02:
So like I said I thought it may be a bad board. Now loading my same build of marlin that I shared 3 days ago that was somewhat working with the uart mode. I have another new board delivered today and all new TMC2209s I get no communication with the drivers??? I downloaded the marlin build from the BigTreeTech repository for the octopus v1.1 and that would not build. I was just going to test your test build on a new board with new drivers bench test. Nothing but errors. I have again exhausted researching on line with no solution. May I ask using bigtree Octopus board with Bigtree TMC drivers do I need to solder jumpers or cut pins?? The first board is now giving me eeprom errors so I think I killed that one. I don't think I have ever had so much troubleshooting on new equipment like this. 

SENDING:M122
axis:pwm_scale/curr_scale/mech_load|flags|warncount
		X	Y	Z	Z2	E
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	650	650	800
RMS current	1436	1436	1160	1160	1436
MAX current	2025	2025	1636	1636	2025
Run current	25/31	25/31	20/31	20/31	25/31
Hold current	12/31	12/31	10/31	10/31	12/31
CS actual	0/31	0/31	0/31	0/31	0/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false	false
msteps		256	256	256	256	256
interp		false	false	false	false	false
tstep		0	0	0	0	0
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false
triggered
 OTP		false	false	false	false	false
pwm scale sum	0	0	0	0	0
pwm scale auto	0	0	0	0	0
pwm offset auto	0	0	0	0	0
pwm grad auto	0	0	0	0	0
off time	0	0	0	0	0
blank time	16	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3	-3
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	0	0	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
sg_result	0	0	0	0	0
stst		*	*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		Z2	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing Z2 connection... Error: All LOW
Testing E connection... Error: All LOW

HELP PLEASE....

radek8, 2021-12-02:
Send a photo of the wired board. Send a photo of tmc2209 from the side of the pins

radek8, 2021-12-02:
Check if the pins for UART mod are connected

[image]
[image]

jmarkscnc, 2021-12-02:
Sorry I figured it out. Very rookie mistake, I have several power leads to supply power on my bench and I had the circuit supplying 24v DC to the steppers was but not switched on. my apologies.
 Driver registers:
		X	0xC0:0C:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0A:00:00
		Z2	0xC0:0A:00:00
		E	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

Now I just can not figure out the Error: Line number is not last Line Number+1 last line 0. I think is is a high low report but I can not find the issue.
I think it may be in the thermistors settings for the bed?
if I switch my screen to Marlin mode I can see that the Hot End shows 0degrees and 0degrees. However the bed shows 0 degrees and err. However I get E0 showing 0 degrees and 0 degrees and the bed showing 0 degrees and 0 degrees between the warning messages.
Just to be clear these settings are all being tested on the new 2nd octopus board.

jmarkscnc, 2021-12-02:
I switch my screen to Marlin mode I can see that the Hot End shows 0degrees and 0degrees. However the bed shows 0 degrees and err.

I get E0 showing 0 degrees and 0 degrees and the bed showing 0 degrees and 0 degrees between the warning messages on the touch screen mode.

jmarkscnc, 2021-12-03:
First thing is I want to thank you for your assistance. I do really appreciate all of you help.

   Ok now that  I am back where I started with the first board before it went to heck. Board 2 is now up and running but the Z  and Z2 will not work exacting a G34 Z align. Also I am back to the problem with the E0 moving in sink with the Z axis. This is what I the first board went crazy and lit up all of the heater mosfets like a Christmas Tree.. I guess no refund or replacement on that board.....

   In Marlin I commented out E0 and uncommented E1. I then moved E0 stepper to motor 5 and also Moved Z2 to motor 4 leaving motor 3 open. Wow everything seems to be working.  Same build except for the TFT rebuild with the fixes from yesterday seems to have done the trick.  

   One thing now is the tft70 when loading filament in touch mode, the stepper runs the unload and then a warning comes up waiting for user input. you can click ok forever and this alarm does go away. Should this not say load or purge more? I just uploaded the latest build to the tft70 and compiled. I also got a language file error to load the language in my case English? 

   I will recompile and see if that fixes the errors with the load/unload. I will let you know. If anyone has any suggestions on this it would help me a lot.  It is late here so I'm calling it a night for today.  Thanks again!

radek8, 2021-12-03:
I hope you noticed that the connector for the Z motor is duplicated?

jmarkscnc, 2021-12-03:
My apologies I miss typed the Motor locations. Yes this is my current connections on the board.

 
[image]

 
[image]

I read on other posts that if moving the E0 to motor 4 you would need to label this E1 in Marlin. I am trying to follow the mapping of the Marlin code but I am just a rookie when I get in too deep in the logic. Could that be the issue with the load/unload?

radek8, 2021-12-03:
If you want to use a different default connector, just edit the pin file for your board.

monsterjeepn, 2022-02-10:
> If you want to use a different default connector, just edit the pin file for your board.

This is the right answer. To expand a bit, there are two spots to modify/verify in the "pins_BTT_OCTOPUS_V1_common.h" file. Search for "X_STEP_PIN" and modify the step/dir/en/cs pin settings according to your hardware configuration. Also search for "X_SERIAL_TX_PIN" and modify the TX/RX pin settings for the TMC drivers in your hardware configuration. The schematic and the board layout (*PINS.pdf) are invaluable to figure this out.