bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #90: Cant control any other fans than single part cooling fan

hololight, 2021-12-05:
Whenever I try to use m106 p* s255, I can only control the part cooling fan (pin 0). Pin 1 is assigned to auto extruder cooling (and kicks on fine above the set 50C). But when I plug a fan into any of the other PWM fan outputs, I can't control them (they never turn on with any value of P* with my M106 command. I'm eventually going to be controlling a Laser, but in the meantime I was testing with just another blower fan.

radek8, 2021-12-05:
You can only control ventilators that you define in FW

hololight, 2021-12-05:
Can you point me to the correct area for me to take a look? I'm just not seeing it... in fans.h there is already 3 fans listed:
#define FAN_PIN                             PE5   // Fan0
#define FAN1_PIN                            PE6   // Fan1
#define FAN2_PIN                            PC8   // Fan2

in configuration.h I'm not seeing where I would define it other than the following (which the comment claims by leaving the line commented out, enables all fans:

// Set number of user-controlled fans. Disable to use all board-defined fans.
// :[1,2,3,4,5,6,7,8]
//#define NUM_M106_FANS 1

hololight, 2021-12-05:
So I got it working. I made two changes (not sure which [or both?] made the difference):

In configuration_adv.h:
I commented out the controller cooling fan (I dont need it since I use an always on fan connected two one of the two ports used for that purpose). My logic was that the cooling fan was defined as using FAN2_PIN, so I could use that pin perhaps, if I stopped it from being used by the cooling fan...

// #define USE_CONTROLLER_FAN

Then in configuration.h:
This is the more likely culprit... I removed the comment and set the number of fans to 4.... according to the documentation I shouldn't' have had to do this, but I think its what got it working.

Now on to hooking up the laser to this new 'dedicated' port

closed 2021-12-05 by hololight