bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #91: USB thumb drive support, Marlin, Octopus 1.1

roschme, 2021-12-13:
Hi,

i'm in trouble compiling Marlin-bugfix-2.0.x with activated " #define USB_FLASH_DRIVE_SUPPORT " and " #define USE_OTG_USB_HOST "for an Octopus 1.1 under VSC Platformio. What i'm doing wrong? Would be very grateful for your hints and help. 
(USB Host Support/Pronterface works as everthing else but the support for usb thumbs fails)  

thx bob

compilerlog:

C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::isReady()':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:266: undefined referenCe to `USBHost::getUsbTaskState()'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:266: undefined referenCe to `usb'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::CardSize()':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:296: undefined referenCe to `bulk' 
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:294: undefined referenCe to `BulkStorage::GetCapaCity(unsigned Char)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::readBloCk(unsigned long, unsigned Char*)':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:309: undefined referenCe to `BulkStorage::Read(unsigned Char, unsigned long, unsigned short, unsigned Char, unsigned Char*)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:299: undefined referenCe to `bulk'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::writeBloCk(unsigned long, unsigned Char Const*)':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:323: undefined referenCe to `BulkStorage::Write(unsigned Char, unsigned long, unsigned short, unsigned Char, unsigned Char Const*)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:313: undefined referenCe to `bulk'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::usbStartup()':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:127: undefined referenCe to `USBHost::start()'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:142: undefined referenCe to `usb'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::idle()':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:151: undefined referenCe to `USBHost::Task()'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:153: undefined referenCe to `USBHost::getUsbTaskState()'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:209: undefined referenCe to `BulkStorage::LUNIsGood(unsigned Char)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:243: undefined referenCe to `BulkStorage::LUNIsGood(unsigned Char)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:254: undefined referenCe to `usb'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:254: undefined referenCe to `bulk'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_OCTOPUS_V1/srC/srC/sd/usb_flashdrive/Sd2Card_FlashDrive.Cpp.o: in funCtion `DiskIODriver_USBFlash::init(unsigned Char, short)':
C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:274: undefined referenCe to `BulkStorage::GetSeCtorSize(unsigned Char)'
C:/users/xyz/.platformio/paCkages/toolChain-gCCarmnoneeabi/bin/../lib/gCC/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.x/Marlin\srC\sd\usb_flashdrive/Sd2Card_FlashDrive.Cpp:285: undefined referenCe to `bulk'

roschme, 2021-12-15:
USB Flash Disk finally working on the Octopus V1.1: 
Just use the ORIGINAL! https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip

platformio.ini:
...
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_OCTOPUS_V1_USB
include_dir  = Marlin  
...

configuration.h:
...
#define CONFIGURATION_H_VERSION 02000902
...
#define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1 
...
#define SDSUPPORT
...

configuration_adv.h :
...
#define USB_FLASH_DRIVE_SUPPORT //bob
...
#define USE_OTG_USB_HOST
...
#define NO_SD_HOST_DRIVE   //Disable SD Card access over USB (for security). (otherwise problems with pronterface under win)

usama0300854, 2024-02-09:
i downloaded Marlin-bugfix-2.1.x.. remember to download bugfix..this file will be larger than the rest.

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F446ZE_btt_usb_flash_drive
include_dir  = Marlin

>>>>> in Configuration.h :

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1
#endif

#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1
#endif

#define SDSUPPORT

>>>  configuration_adv.h :
...
#define USB_FLASH_DRIVE_SUPPORT
...
#define USE_OTG_USB_HOST
...
#define NO_SD_HOST_DRIVE //Disable SD Card access over USB (for security). (otherwise problems with pronterface under win)