bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #94: HE0-HE3 are always on

Wukii2012, 2022-01-14:
So I have a new board that I setup with the firmware powered via USB.  When I changed over to normal power (took out the jumper) the board powers on and seems to be working fine however the HE0-HE3 are also powered on and showing a green light next to them. They started to heat up my bed and hotend. Should this be happening??

qcgilbert, 2022-02-05:
I am experiencing the same issue

jjkrygsman, 2022-10-08:
I'm experiencing the same issue on multiple boards, even with Klipper firmware.  HE0-HE3 eventually turn off, but not sure why they are on the begin with.

the1snm, 2022-10-29:
I've got 2 octopus pro boards and 1 does not have this problem and the other does. Nothing seems to turn off these heaters. I've swapped out my steppers, reflashed the board, removed all the wires from the board just 24v hooked up and still the heaters turn on. Aside from the heaters on, everything else works perfectly, all motors move, endstops, fans etc. I'm still inside my return window so I've got a replacement on it's way, don't know what else to test.

ajenkuhnabat, 2024-03-06:
Did you see the pin changes here: https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro 
They swapped Motor4 ENable and HE0, would would case this issue. I ran ito this by myself and solved it by correcting the klipper config pins.