bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #99: Erro compiling MArlin Bugfix 2.0.9.2 on a Bigteeetech Octpus V1.1

Delanjp949, 2022-02-05:
Getting the following error con compile,
Whta am I doing wrong. Cannot see the error. 
Compiling using Visual Studio.

help apprectated.

James

> Executing task in folder Marlin-bugfix-2.0.9.2.x: C:\Users\james\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_OCTOPUS_V1 (platform: ststm32@~12.1; board: marlin_BigTree_Octopus_v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 79486: character maps to <undefined>:
  File "C:\Users\james\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 179:
    env.SConscript(item, exports="env")
  File "C:\Users\james\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\james\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\james\Download\BIGTREETECH-OCTOPUS-V1.0-master\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.2.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 253:
    compute_build_signature(env)
  File "C:\Users\james\Download\BIGTREETECH-OCTOPUS-V1.0-master\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.2.x\buildroot\share\PlatformIO\scripts\signature.py", line 83:
    defines = extract_defines(header)
  File "C:\Users\james\Download\BIGTREETECH-OCTOPUS-V1.0-master\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.2.x\buildroot\share\PlatformIO\scripts\signature.py", line 15:
    f = open(filepath).read().split("\n")
  File "C:\Users\james\.platformio\python3\lib\encodings\cp1252.py", line 23:
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
====================================================================================== [FAILED] Took 1.32 seconds ======================================================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_OCTOPUS_V1  FAILED    00:00:01.320
================================================================================= 1 failed, 0 succeeded in 00:00:01.320 =================================================================================The terminal process "C:\Users\james\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

monsterjeepn, 2022-02-08:
I encountered this as well. [Marlin fix #23303](https://github.com/MarlinFirmware/Marlin/pull/23303/files) worked for me.

Delanjp949, 2022-02-08:
All fixed thanks.  Deleted the .Plantformio folder under the user profile.  Force a reintsll on PIO and all worked afterwards

closed 2022-02-08 by Delanjp949