bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #101: error while compiling

Tim9557, 2022-02-07:
[{
	"resource": "/c:/Users/Tim/Downloads/BIGTREETECH-OCTOPUS-V1.0-master/BIGTREETECH-OCTOPUS-V1.0-master/Firmware/Marlin-bugfix-2.0.9.2.x/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\Tim\\Downloads\\BIGTREETECH-OCTOPUS-V1.0-master\\BIGTREETECH-OCTOPUS-V1.0-master\\Firmware\\Marlin-bugfix-2.0.9.2.x",
	"severity": 4,
	"message": "Cannot find \"C:\\Users\\Tim\\.platformio\\packages\\framework-arduinoststm32\\variants\\MARLIN_BIGTREE_OCTOPUS_V1\".",
	"startLineNumber": 39,
	"startColumn": 17,
	"endLineNumber": 39,
	"endColumn": 112
},{
	"resource": "/c:/Users/Tim/Downloads/BIGTREETECH-OCTOPUS-V1.0-master/BIGTREETECH-OCTOPUS-V1.0-master/Firmware/Marlin-bugfix-2.0.9.2.x/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\Tim\\Downloads\\BIGTREETECH-OCTOPUS-V1.0-master\\BIGTREETECH-OCTOPUS-V1.0-master\\Firmware\\Marlin-bugfix-2.0.9.2.x",
	"severity": 4,
	"message": "Cannot find \"C:\\Users\\Tim\\.platformio\\packages\\framework-arduinoststm32\\variants\\MARLIN_BIGTREE_OCTOPUS_V1\".",
	"startLineNumber": 77,
	"startColumn": 21,
	"endLineNumber": 77,
	"endColumn": 116
},{
	"resource": "/c:/Users/Tim/Downloads/BIGTREETECH-OCTOPUS-V1.0-master/BIGTREETECH-OCTOPUS-V1.0-master/Firmware/Marlin-bugfix-2.0.9.2.x/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\Tim\\Downloads\\BIGTREETECH-OCTOPUS-V1.0-master\\BIGTREETECH-OCTOPUS-V1.0-master\\Firmware\\Marlin-bugfix-2.0.9.2.x",
	"severity": 4,
	"message": "Cannot find \"arm-none-eabi-gcc\".",
	"startLineNumber": 107,
	"startColumn": 29,
	"endLineNumber": 107,
	"endColumn": 48
}]

pilgerone, 2022-02-27:
Same thing here. Can't compile. Have you manage to find a workaround?

Hman32, 2022-03-27:
@pilgerone  can you try to remove the .vscode folder and start again your VS Code?
Normally they should not include the .vscode folder into the repository :/

pilgerone, 2022-04-03:
> @pilgerone can you try to remove the .vscode folder and start again your VS Code? Normally they should not include the .vscode folder into the repository :/

I actually ended up taking the vanilla Marlin from the original repo and adjusting it to the octopus. I took the config files from this one and compared with the original stuff, changing what needed to be changed. Took quite some work, but it's working flawlessly.