bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #103: Board not working when Jumpers were not kept

selvakumaran2016, 2022-02-16:
Hi all,

We are using the Octopus V1.1 board. We have already used 2-3 boards and all are working properly.

In one board, while keeping the jumpers for the Motor 7 Stepper driver, we forget to keep any jumpers. (All other 6 motors are properly connected with Jumpers). Without any jumper, we inserted A4988.

A small burning smell came and the board became dead. 

The marked chip in the below image looks burnt. The main processor is not showing any signs of burning.

Please suggest whether we can change this? and what is that IC name?

Thanks in advance

[image]

radek8, 2022-02-20:
It should be a 3.3V VCC voltage regulator

[image]

radek8, 2022-02-20:
These should be two matched voltage regulators.

[image]

radek8, 2022-02-20:
RT9193-33GB

radek8, 2022-02-20:
https://de.aliexpress.com/item/4000387137481.html?gatewayAdapt=glo2deu&af=3762815_1&cv=37665065&cn=45r7m8frcmtfz2dbql3r2t7bbcrgho48&dp=v5_45r7m8frcmtfz2dbql3r2t7bbcrgho48&utm_source=epn&utm_medium=cpa&utm_campaign=3762815_1&utm_content=37665065&product_id=4000387137481&afref=https%3A%2F%2Falipromo.com%2Fredirect%2Fcpa%2Fo%2Fq3uk36t2i4ehtdo7wys72lakq0y01lgl%2F&aff_fcid=1239808ea90941e28f5889b97f3c29f1-1645381431820-06490-_d6yGheF&aff_fsk=_d6yGheF&aff_platform=api&sk=_d6yGheF&aff_trace_key=1239808ea90941e28f5889b97f3c29f1-1645381431820-06490-_d6yGheF&terminal_id=290428c8106c48438ea1ffecefca338b

selvakumaran2016, 2022-02-21:
@radek8 Great. Thanks for the support. Will order this...

closed 2022-02-21 by selvakumaran2016