bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #106: FSMC display support

mkornmue, 2022-02-22:
I want to use an FSMC touch display with an ILI9486 Driver but there is no information how to connect the display for that.

Is FSMC supported for the Octopus V1.1?

I can find FSMC ports for the STM32F446 on EXP1 so it should be possible. Am I right?

TheCount2255, 2022-02-22:
You would need to trace out the EXP1 pins to the pins on the back of the display. Once you have the cabling correct you should be good to go.

mkornmue, 2022-02-23:
The thing is why I'am struggeling is that on the STM website there is no information regarding FSMC for this specific controller (STM32F446 ). The earliest version with FSMC is STM32F429.

So could it be that STM32F446 is really not supporting it?