bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #107: Z2 stepper not responding - Octopus Pro V1.0 - Marlin 2.0.9.3

TheCount2255, 2022-02-22:
Having an issue getting Z2 to respond to M122 command. I have tried a lot of diagnostics on this and can't find a proper solution.

Physical board setup as follows:
Stepper X,Y,Z(1),Z2,E0 are setup in Motor 0-4 plugs.

Marlin setup: 
- Conf.h 
```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
```

-Conf_adv.h
```
//
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  // Enable if Z motor direction signals are the opposite of Z1
  //#define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR
```

- ...Octopus_V1_common.h (pins file)
```
//
// Steppers
//
#define X_STEP_PIN                          PF13  // MOTOR 0
#define X_DIR_PIN                            PF12
#define X_ENABLE_PIN                     PF14
#ifndef X_CS_PIN
  #define X_CS_PIN                           PC4
#endif

#define Y_STEP_PIN                         PG0   // MOTOR 1
#define Y_DIR_PIN                           PG1
#define Y_ENABLE_PIN                    PF15
#ifndef Y_CS_PIN
  #define Y_CS_PIN                          PD11
#endif

#define Z_STEP_PIN                         PF11  // MOTOR 2
#define Z_DIR_PIN                           PG3
#define Z_ENABLE_PIN                    PG5
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PC6
#endif

#define Z2_STEP_PIN                       PG4   // MOTOR 3
#define Z2_DIR_PIN                         PC1
#define Z2_ENABLE_PIN                  PA0
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                        PC7
#endif

#define E0_STEP_PIN                       PF9   // MOTOR 4 
#define E0_DIR_PIN                         PF10
#define E0_ENABLE_PIN                  PG2
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PF2
#endif
```

After hooking the board up to power and usb I send a M122 command and receive the following reply (truncated for brevity)
```
READ: Driver registers:
READ: 		X	0xC0:1F:00:00
READ: 		Y	0xC0:1E:00:00
READ: 		Z	0xC0:1F:00:00
READ: 		Z2	0x00:00:00:00	 Bad response!
READ: 		E	0xC0:1F:00:00
READ: 
READ: 
READ: Testing X connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing Z2 connection... Error: All LOW
READ: Testing E connection... OK
```
After this I have tried moving motor 3 stepper to a different location (motor 5-8) on the board with no luck as expected. I tried a different/new stepper driver(s) to see if this was an issue (No change). 
Thinking that the plug on the board for the stepper might be dead I changed the ...Common.h pins file to reflect a different location (motor 5) and moved the driver to that location. This yielded the same result as above. The last thing that I tried was reversing the dual Z steppers to just 1: 
`#define NUM_Z_STEPPER_DRIVERS 1   // (1-4) Z options change based on how many`
and assigned dual E drivers 
`#define EXTRUDERS 2`
and assign the pins from Z2 to E1
```
#define E1_STEP_PIN                         PG4   // MOTOR 5
#define E1_DIR_PIN                          PC1
#define E1_ENABLE_PIN                       PA0
#ifndef E1_CS_PIN 
#define E1_CS_PIN                         PC7
#endif
```
 with the following results: (Physical the board is setup to motor 0-2,4-5)
```
READ: Driver registers:
READ: 		X	0xC0:0C:00:00
READ: 		Y	0xC0:0C:00:00
READ: 		Z	0xC0:0C:00:00
READ: 		E	0xC0:0C:00:00
READ: 		E1	0xC0:0C:00:00
READ: 
READ: 
READ: Testing X connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK
READ: Testing E1 connection... OK
```
I left the stepper in motor 5 plug reassigned the pins to the normal Z2 pins and got the following error again:
```
READ: 		Z2	0x00:00:00:00	 Bad response!
READ: Testing Z2 connection... Error: All LOW
```

I moved the motor 5 stepper to motor 3 and retested and got the error again.

Things I can safely assume at this point:
- All steppers are working and will report w/ M122
- plugs 0-2,4-5 work w/ steppers plugged in

Things I question:
- Does plug 3 (motor 3) on the board work at? Not sure how to test for this now.
- How/why does using normal Z2 pin definition work on motor 5 when defined as an extruder motor *only*?
- Code issue NOT H/W?

I think* that I can assume that the board is fine, but the motor 3 plug is not addressed correctly/at all. Trying to trace the pins through the code and the automatic pin assignment code is a nightmare for me (not a gooder coder), so would LOVE any and all help.
I'm open to any and all suggestions on a solution (even that I might have a bad board).

ANT-NONIUS, 2022-04-04:
Cleaning driver's pins and sockets with contact cleaner did the trick for me.

kursorlinde, 2022-05-04:
Hi just wanted to tell you i am having the same issue on BTT Octopus Pro V1.0 with TMC2209
tried the same steps as you described with the same results then discovered this thread and the one on reddit

my uneducated guess is that the issue comes from BTT actually paralleling Z0 and Z2 so that "NUM_Z_STEPPER_DRIVERS 1" basically already is equal to 2 however this seems to be a marlin issue after all that you and i have tried, can't think of anything else to try to fix this at this moment.

kursorlinde, 2022-05-07:
looks like i found the mistake for my part 
G34 does work now without errors
MOTOR2 ("THE" Z-Axis Driver) simply has two JST connectors -_-
its not "THE" Z2 thats where i made my initial false assumption

if you want to use Z2 for actually having one individual driver for each motor you have to plug your second Z stepper motor plug into the MOTOR3 (DRIVER3) connector not the MOTOR2_2 one ( MOTOR 2_1 and MOTOR2_2 are both using the DRIVER2 (see PINS pdf file from BTT and pins config file in marlin )

if you try to use dual z with MOTOR2_1 and MOTOR2_2 it doesnt work because for marlin it's only one MOTOR and it is missing the second Z motor 
also i cleaned the connectors because of what ANT-NONIUS suggested on April 4th and March 31st

Shamoon78, 2022-05-16:
Marlin can use many z many y and many x and many extruders as much as you want limited only by what number of drivers motherboard can offer, Iam not following up for the last year and half and forget marlin compyling but i remember I configured marlin to run 3 z stepper indvidually to level the bed and swap x driver for extruder and vice versa but there is configuration how to assign every driver of motor to certain function ,even i had part fan mosfet died so i configured e1 hotend fan connector to run as part fan( hotend fan is fixed speed fan while part fan is controllable speed fan), I think is id number for certain pin connector that the mcu handle have to be changed to the new function(other pin on the board ), I am trying to refresh my information and will be back to explain when i remember.

Shamoon78, 2022-05-16:
Mcu have many data pin numbers are assigned on the board to certain connectors but you can assign these numbers to different connectors or swap between each others.

Arnie3d, 2023-01-15:
what happend can some give me a good solution to Z2 and Y2 not working but M122 wth Pronterface telling me eveything is ok x,y,y2,z,z2 and e0 read OK but no movementon y2 and z2

badonj001, 2023-03-19:
> looks like i found the mistake for my part G34 does work now without errors MOTOR2 ("THE" Z-Axis Driver) simply has two JST connectors -_- its not "THE" Z2 thats where i made my initial false [...]

Same here on that second Z-JST connector. Had me scratching my head until I saw your post, lol

Thanks, lol