bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #108: Unable to connect to octopus pro board

stewcop23, 2022-02-23:
I'm using Octopus pro v1.0, Marlin 2.0.9.3, and windows. I have been successful in uploading firmware but when it comes to connecting the board to my computer (I'm using pronterface) it won't connect. Before you give me the same responses that I've been getting from everyone on the internet.

Yes I have set serial port 2 to be -1
Yes I have tried to update drivers (couldn't really find them though)
Nothing shows up on the device manager

I'm probably doing one small thing wrong, but any help is appreciated

stewcop23, 2022-02-24:
My configuration.h is [Here](https://drive.google.com/file/d/1ARPrd58psREHDHCngPR04Sn8ix2347EH/view?usp=sharing)

Nico4589, 2022-04-15:
I have the same problem

Royalblu, 2022-06-06:
I’ve the same issue. I don’t see a COM port at all in Windows 10 so not selectable in Pronterface. Has anyone resolve this promblem?

BOARD is an Octopus Pro with Marlin 2.1 and 2 x serial ports enabled as follows;

Serial 1 = 1
Serial 2 = -1

Sheridan-Tech, 2022-10-11:
Same issue here

stewcop23, 2022-10-11:
I would recommend contacting BIQU's customer support. There is a significant amount of hoops to jump through, but you may be able to get some sort of refund or discount on your next purchase. 

I apparently missed my warranty period and was only able to get 10% off my next purchase, so i encourage you to email them earlier than i did

Sheridan-Tech, 2022-10-11:
With a lot of debugging effort I was able to get it working. Apparently it was throwing a min temp error code that caused it to halt, thus disabling serial. They should really have beep or flash codes to report board errors back to the user.

Ciceronic, 2023-03-11:
> With a lot of debugging effort I was able to get it working. Apparently it was throwing a min temp error code that caused it to halt, thus disabling serial. They should really have beep or flash codes to report board errors back to the user.

Interesting.
I have the same problem, what did you do to solve the temp error code?

Sheridan-Tech, 2023-04-03:
It was too far back for me to remember exactly, but based on my commits
around that time I think it was related to using the wrong
thermistor definition. I am using a PT100, so it was out of the expected
range.

I changed
#define TEMP_SENSOR_0 1
to
#define TEMP_SENSOR_0 -5

filsh, 2024-05-15:
Some problem, how to fix it?

looxonline, 2024-05-16:
> Some problem, how to fix it?

This is almost 100% not a hardware issue and rather an issue with your firmware config or the config of the serial link on your communication device. I would recommend asking on a discord group such as the Voron group or on the BTT official facebook group where you will find many who can assist you with the firmware and serial config.

closed 2024-05-16 by looxonline