bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #109: ESP8266, ESP-07S Setup Instructions For Marlin Needed

Aurawin, 2022-02-25:
Can someone point me to the instructions to setup my ESP8266 ESP-07S with my V1.1 board?

WesMcGee68, 2022-07-21:
install of the BTT ESP32 install on Octopus v1.1 needed also....

lukasloetkolben, 2022-09-03:
have you found a way to use  WiFi ESP8266 ESP-07S with the V1.1 board?

lukasloetkolben, 2022-09-04:
Ok ESP3D is working for me.
- Connect the ESP07S to a USB to TTL Adapter:
```
TXD0 -> RX
RXD0 -> TX
VCC -> 3.3V
GND -> GND
RST -> 3.3V
EN -> 3.3V
IO0 -> GND
IO2 -> 3.3V
IO15 -> GND
```

Download VS Code and install PlatformIO compile it and flash it to the ESP07S.
If its not possible to flash with PlatformIO use ESP8266 DOWNLOAD TOOL with this config:
```
CrystalFeq: 24M
SPI SPEED: 40MHz
SPI MODE: QIO
FLASH SIZE: 32Mbit
```

follow the instructions: https://github.com/luc-github/ESP3D

[image]

Aurawin, 2022-09-04:
The module needs to be flashed before you can use it.  And before you can flash it you need to configure it.  The WiFi is too slow to use for uploading large prints.  Consider all the new options like Manta 4/8P with CB1.

lukasloetkolben, 2022-09-05:
> The module needs to be flashed before you can use it. And before you can flash it you need to configure it. The WiFi is too slow to use for uploading large prints. Consider all the new options like Manta 4/8P with CB1.

The upload is not slow because of the wifi, it is because of the communication between ESP and Marlin.

WesMcGee68, 2022-10-10:
Yes I did get it worked out..

Erfan017, 2023-09-29:
you can use this for esp32u also
i used arduino uno as a programmer and programmed my esp32u for my octopus pro