bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #112: BTT Ocotopus v1.1 - Noctua NF-A6x25 FLX Fan Don't spin!

MihaiCiornei, 2022-03-09:
I recently upgraded the CR-10S with Octopus v1.1., Due to the fact that the old board made the engines make a hell of a noise. Everything is perfect. I also installed some NTC 3950 100 K ohm, to make a fan start when the Stepper Drive Y reaches a temperature of 40, one on the power supply for rasbery Pi, everything works with Noctua NF-A4x20 FLX fans. They turn on and off exactly as they should ... but when I wanted to replace these fans with a Noctua NF-A6x25 FLX, it didn't spin. I'm sure it works, I checked it by replacing the power supply fan. Do you have any idea what the problem is?

This is my cfg:

```
# This file contains common pin mappings for the BigTreeTech Octopus.

# To use this config, the firmware should be compiled for the

# STM32F446 with a "32KiB bootloader" and a "12MHz crystal" clock reference.

# See docs/Config_Reference.md for a description of parameters.

# Driver0

[stepper_x]

step_pin: PF13

dir_pin: !PF12

enable_pin: !PF14

microsteps: 16

rotation_distance: 40

endstop_pin: PG6

position_endstop: 0

position_max: 280

homing_speed: 50

# Driver1

[stepper_y]

step_pin: PG0

dir_pin: !PG1

enable_pin: !PF15

microsteps: 16

rotation_distance: 40

endstop_pin: PG9

position_endstop: 0

position_max: 280

homing_speed: 50

# Driver2

[stepper_z]

step_pin: PF11

dir_pin: PG3

enable_pin: !PG5

microsteps: 16

rotation_distance: 8

endstop_pin: PG10

position_endstop: 0.5

position_max: 300

# Driver4

[extruder]

step_pin: PF9

dir_pin: !PF10

enable_pin: !PG2

microsteps: 16

rotation_distance: 29.7

nozzle_diameter: 0.400

pressure_advance = 0.560

filament_diameter: 1.750

heater_pin: PA2 # HE0

sensor_pin: PF4 # T0

sensor_type: Generic 3950

min_temp: 0

max_temp: 220

[heater_bed]

heater_pin: PA1

sensor_pin: PF3 # TB

sensor_type: Generic 3950

min_temp: 0

max_temp: 130

[fan]

pin: PA8

kick_start_time: 0.5

off_below: 0.10

[heater_fan hotend_fan]

pin: PE5

max_power: 1.0

kick_start_time: 0.5

heater: extruder

heater_temp: 50.0

[temperature_fan fan_Stepper_Drivers_Y]

pin: PD12

max_power: 1.0

sensor_type: Generic 3950

sensor_pin: PF5

min_temp: 0

max_temp: 60

target_temp: 40.0

max_speed: 1.0

min_speed: 0.3

control: watermark

gcode_id: stepper_y

[temperature_fan fan_CASE]

pin: PD13

max_power: 1.0

sensor_type: Generic 3950

sensor_pin: PF6

min_temp: 0

max_temp: 60

target_temp:40.0

max_speed: 1.0

min_speed: 0.3

control: watermark

gcode_id: alim_rasbery

[mcu]

serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_200004000450534E4E313120-if00

[printer]

kinematics: cartesian

max_velocity: 300

max_accel: 3000

max_z_velocity: 5

max_z_accel: 100

########################################

# TMC2209 configuration

########################################

[tmc2209 stepper_x]

uart_pin: PC4

diag_pin: PG6

run_current: 0.800

stealthchop_threshold: 999999

[tmc2209 stepper_y]

uart_pin: PD11

diag_pin: PG9

run_current: 0.800

stealthchop_threshold: 999999

[tmc2209 stepper_z]

uart_pin: PC6

diag_pin: PG10

run_current: 0.650

stealthchop_threshold: 999999

[tmc2209 extruder]

uart_pin: PF2

run_current: 0.800

stealthchop_threshold: 999999

[board_pins]

aliases:

# EXP1 header

EXP1_1=PE8, EXP1_2=PE7,

EXP1_3=PE9, EXP1_4=PE10,

EXP1_5=PE12, EXP1_6=PE13, # Slot in the socket on this side

EXP1_7=PE14, EXP1_8=PE15,

EXP1_9=<GND>, EXP1_10=<5V>,

# EXP2 header

EXP2_1=PA6, EXP2_2=PA5,

EXP2_3=PB1, EXP2_4=PA4,

EXP2_5=PB2, EXP2_6=PA7, # Slot in the socket on this side

EXP2_7=PC15, EXP2_8=<RST>,

EXP2_9=<GND>, EXP2_10=PC5

#*# <---------------------- SAVE_CONFIG ---------------------->

#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.

#*#

#*# [extruder]

#*# control = pid

#*# pid_kp = 24.604

#*# pid_ki = 1.233

#*# pid_kd = 122.713

#*#

#*# [heater_bed]

#*# control = pid

#*# pid_kp = 71.732

#*# pid_ki = 0.968

#*# pid_kd = 1328.834
```

MihaiCiornei, 2022-03-13:
Rly .... no response :(

Hman32, 2022-04-16:
Did you check the voltage output and the polarity on the fan?
You could try to invert the Gnd/V+ pin from the fan and check if it's the issue.

MihaiCiornei, 2022-04-17:
Thank you very much, I will try, at the first opportunity when I disassemble the printer, it is not a bad idea.