bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #114: Octopus v1.1 major hardware issue consistent across both boards causing frequent motor malfunctions

zvonka, 2022-04-03:
I have been a fan of BTT products for many years, everything I owned that was BTT before worked flawlessly but after now three seperate boards im realizing there is a grave problem with the board design at a hardware level

thankfully it is something that is very easily fixed!

While this might not fix your motor issues, it worked for all three of my boards after I applied this fix & I've been satisfied since with the performance.

My suspicion is that the stepper driver sockets are slightly too close together, causing somehow voltage to stray from one driver to the next causing all kinds of problems. I put a thin sheet of kapton tape in between each stepper driver before plugging into the socket all the way to isolate them from each other & surprise the motor issues all went away.

Worth a shot if anyone is having problems still, good luck all.

ANT-NONIUS, 2022-04-03:
Hi, having similar problems with steper drivers. Your "theory" has a lot of sense.
I worked around the issue not fully pushing the drivers on the sockets.
I was blamming flux contamination on the sockets. But a closser look, after reading your comment, make me realize that I was putting them at slightly different heights, therefor not making contact between circuits.

ANT-NONIUS, 2022-04-04:
Cleaning all the pins and sockets with contact cleaner did the trick for me.

tanguy-lq, 2022-04-28:
Hi, i can confirm the issue. My drivers were touching eachother and my motors were doing weid stuff. The kapton fix works perfectly.
Many thanks to you zvonka, i was searching for quite a while now !

zvonka, 2022-07-21:
Pin extensions might work but I can imagine there would be some miniscule voltage drop due to additional number of interconnections that electricity would have to travel through. 

If you don't have kapton tape, you could easily use just a very thin sheet of card stock. Like index-card paper cut down small enough to fit in between each of the stepper drivers. I have TMC2209 and they never get even warm when printing on my second printer with the same board & I used just a piece of paper on that one instead of kapton tape & it resolved the abnormal movement issues I was having.

CircuitSetup, 2022-08-03:
Interesting. This must be happening because of excessive ESD/crosstalk coming from the drivers. I wonder if the Octopus or step stick pcb's need some more decoupling caps to gnd. Are the wires for the motors physically close to one another? Are they twisted together at all?

In the [tmc2209 datasheet (page 14)](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf), it mentions 2 methods for handling ESD. The [step sticks do not have either](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf). [Nor does the Octopus.](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/Hardware/BIGTREETECH%20Octopus.pdf)

Simply grounding the stepper motors individually to DC gnd may have the same effect of discharging ESD.

ArianNasser, 2025-06-27:
Hey I know this thread is old, but do you or anyone have an image of the setup with the tape?