bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #117: power led and status led not working, port not shown. help me out of this

titansmu2, 2022-05-06:


titansmu2, 2022-05-06:
Everything looking good, nothing look like burnt. If i try power chip externally power led and status led are working but port not shown. And while powering the chip by external, chip become get hot. Help me out of this problem
[image]

antwal, 2022-10-16:
i have a similar problem, the card worked smoothly for 15 days, then it died giving me the same problem.

The status led is on but power is always off, it does not work with the USB-C power supply nor with the external power supply.

In the end I checked and it does not deliver 3.3v, but only 5v. In the end I bought a new card hoping it will last longer than the first one