bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #120: More problems than its worth!!

Cabinfourus, 2022-06-15:
The firmware BTT provides is not working anymore. 
I downloaded the firmware-F446-USB.bin, renamed it to firmware.bin and flashed my Octopus 1.0 board but Klipper keeps telling me I must be running an old version of firmware and I need to reflash my board. 
Has Klipper updated their side and BTT didn't?

Cabinfourus, 2022-06-18:
Klipper has been updated. So this means the pre-configured firmware BTT provides in this repository is no longer working.

Hman32, 2022-11-18:
@zvonka what issue did you get with Marlin? I've moved from the BTT 2.0.9 to the official marlin bugfix and got no issue.
Just needed to merge my previous config foles with the new structure/change on Marlin 2.1