bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #123: install of the BTT ESP32 install on Octopus v1.1 needed also....

WesMcGee68, 2022-07-24:
install of the BTT ESP32 install on Octopus v1.1 needed also....

_Originally posted by @WesMcGee68 in https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/109#issuecomment-1191590801_

lukasloetkolben, 2022-08-31:
Had you found a way to connect ESP to your Octopus Board?

Hman32, 2022-09-11:
Same for me, I've buy the BTT ESP32U and there's no documentation on how to install it.

lukasloetkolben, 2022-09-11:
> Same for me, I've buy the BTT ESP32U and there's no documentation on how to install it.

I got it working:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/109#issuecomment-1236359375

Hman32, 2022-09-12:
Hi @lukasloetkolben  could you describe all the step needed to compile and  flash it ?
I've already buy the BTT Writer but I must look how I can connect it to the ESP32 
The documentation from BTT is nearly inexistent :(

aaronse, 2023-01-26:
Also trying to get BTT ESP32U module (purchased from BIQU) working on my Octopus v1.1.  This is for my V1 Engineering Lowrider 3 CNC.  More details available at https://forum.v1e.com/t/firmware-for-bigtreetech-esp32u-esp32e-module-for-octopus-or-skr3-skr-3-ez-skr-2/36403

My current plan is to use my ESP01s programmer with bunch of dupont wiring to the ESP32U module based on the ESP-07S pinout, and try to flash ESP3D compiled to ESP32 firmware. Wanted to check in here incase someone’s been down this path already?  

Wondering if there's an easier way.  For example I stumbled onto (last update 2022/9/6) https://github.com/bigtreetech/BTT-Expansion-module/blob/master/ESP-WIFI/ESP%20WIFI%20Module%20Instruction.pdf , so maybe there's a way to update ESP32U module firmware via SD card reader on the mother board, Octopus in this case.  Didn't see anything in Octopus User Guide https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/BIGTREETECH_Octopus_EN_updated_0719.pdf

WesMcGee68, 2023-01-29:
I did get my Esp32 working that same way.  It was fairly easy with platform io