bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #124: Unable to update Octopus v1 bootloader

dsplendido, 2022-07-27:
Hi, I'm unable to connect to the board and update the bootloader via STM32 programmer, I get the error "no debug probe detected". I'm trying to run Reprap on the octopus and the board chip is STM32F446ZE

12Testing3, 2022-08-05:
did you follow gadgets angels guide?

dsplendido, 2022-08-05:
Not familiar with that, can you post a link? I followed BTT's guide but that left me stuck at the debug probe error. I should preface that I do little to no "programming" per se.

12Testing3, 2022-08-05:
https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/bootloader_bin/backed_up_original_bootloaders/Octopus%20V1.1/Guide%20ST-LINK%20Utility%20on%20Windows%2010%20for%20ST-Link-V2-Clones%20for%20accessing%20STM32%20Processors.pdf