bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #126: BigTreeTech TFT TouchScreen Touchscreen mode possible? Or add that mode please.

Trackhe, 2022-08-10:
BigTreeTech TFT TouchScreen Touchscreen mode possible? Or add that mode please.

When i want to run the BT TFT with my Octopus v1.1 and Klipper i only find to use that with

```
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: PA6
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
en_pin: EXP1_4
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1
```

but this is buggy and ugly. i dont want to send this nice display back and buy another one for the raspberry. for klipperscreen.

So i want a solution to use the touchmode with klipper.

Hman32, 2022-09-11:
Which TFT are you using, the TFT35? there so many at BTT...

Could be the issue is with Klipper:
https://github.com/Klipper3d/klipper/issues/2546

Trackhe, 2022-09-11:
Jea clearly, i m talking about the [BIGTREETECH TFT28 V3.0 & TFT43 V3.0 & TFT50 V3.0 & TFT70 V3.0 Displays](https://biqu.equipment/collections/lcd/products/btt-tft43-v3-0-tft50-v3-0-tft70-v3-0-display-touch-screen-two-working-modes)