bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #129: No USB Connection possible

cwagenl, 2022-08-30:
I have taken the marlin Firmware from this repository, compiled it with Visual Studio Code and uploaded it via SD card. The firmware file is sucessfully renamed to firmware.cur.

The is not connected to any periphials, except a riprap discount display and a 24v power supply. The power jumper is disconnected, so the board does not start up upon usb connection. There are no steppers installed, but their jumpers are set to the TBC2209 setting. 

On my Windows 10 PC, the STM Cube Programmer Software has successfully been installed. 

Here comes the issue: If power up the board after the firmware update and plug the board into the PC with the cable delivered with the board, there is no additional COM-port detected. Therefore, I cannot connect to the board. The issue remains in the following scenarios:
- same PC, Board powered up via USB with the power-jumper set and the 24v power supply off
- same PC, Board powered up via USB with the power-jumper disconnected and the 24v power supply off
- same PC, Board powered up via 24v power supply with the power-jumper disconnected and the 24v power supply on
- Different Windows 10 PC w/o the drivers from STM installed, Board powered up via 24v power supply with the power-jumper disconnected and the 24v power supply on

The firmware used is exactly the one distributed here with the configuration.h-file unchanged.

In all cases, no COM-port is being added to the device manager and there is no connection sound when plugging the board in.

Appreciate any help!

Hman32, 2022-09-11:
Did you compiled the enviroment version that end with "btt_USB" ?

mlee12382, 2022-09-20:
Flip the usb C end over 180degrees and try again, also try a different cable.

Patm91, 2022-11-16:
Was this issue ever resolved?

jgbby, 2025-03-05:
^^, would love to know