bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #131: Raspberry Pi J26 no serial connectivity (Marlin)

westhedge, 2022-09-18:
I have tried everything I can think of, but I cannot get this board to communicate to a Raspberry Pi over the J26 PD5/PD6 pins. I have jumpered them to the GPIO 14/15 pins on the Pi (tx to rx and rx to tx) and I have tried:
#define SERIAL_PORT_3 2
#define BAUDRATE_3 115200

and
#define SERIAL_PORT_3 3

Right now I have SERIAL_PORT -1 (for a USB connection) and SERIAL_PORT_2 1 (for the TFT screen)

But it is not working. What is the correct setting for this or is it just a feature that doesn't work? I can't find any useful information online about this anywhere

westhedge, 2022-09-18:
I discovered my issue.
First: these lines are must be in Marlin:
#define SERIAL_PORT_3 2
#define BAUDRATE_3 115200

Next, you must connect with /dev/TTYAMA0. I had mistakenly connected with with /dev/TTYS0.
The board now works correctly!

Hman32, 2022-10-15:
Thanks for your research and  advice @westhedge