bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #133: COmpilation issue

DaLiV, 2022-10-26:
# installed stm32duino library
arduino-cli config add board_manager.additional_urls https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
arduino-cli update
arduino-cli core install STMicroelectronics:stm32

# defined build options for Octopus Pro on F429ZGT
STMOPTS='--build-property build.series=STM32F4xx'
STMOPTS+=' --build-property build.product_line=STM32F429xx'
STMOPTS+=' --build-property build.board=GENERIC_F429ZGTX'
STMOPTS+=' --build-property build.variant.path=''/home/user/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32F4xx/F427Z(G-I)T_F429ZET_F429Z(G-I)(T-Y)_F437Z(G-I)T_F439Z(G-I)(T-Y)/'''
STMOPTS+=' --build-property build.variant=''STM32F4xx/F427Z(G-I)T_F429ZET_F429Z(G-I)(T-Y)_F437Z(G-I)T_F439Z(G-I)(T-Y)'''
STMOPTS+=' --build-property build.variant_h=variant_generic.h'
STMOPTS+=' --build-property upload.maximum_size=1048576'
STMOPTS+=' --build-property upload.maximum_data_size=196608'
# starting commpilation
arduino-cli compile -b STMicroelectronics:stm32:GenF4 $STMOPTS --verbose

-------
result : 
/home/user/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/stm32_def.h:64:17: error: the value of 'SystemCoreClock' is not usable in a constant expression
   64 |   #define F_CPU SystemCoreClock
      |                 ^~~~~~~~~~~~~~~
/home/3d/BTT-Octo/Firmware/Marlin-bugfix-2.0.9.3.x/Marlin/src/HAL/STM32/HAL.cpp:71:27: note: in expansion of macro 'F_CPU'
   71 |   constexpr int cpuFreq = F_CPU;
      |                           ^~~~~
...
/home/user/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/system_stm32f4xx.h:57:17: note: 'uint32_t SystemCoreClock' is not const
   57 | extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */
      |                 ^~~~~~~~~~~~~~~

==============
at the end sketch is not compiled.

DaLiV, 2022-10-26:
additionaly : 
firstly was error about serial port
- commented in "Configuration.h"
// #define SERIAL_PORT_2 -1

other settings at this point was untouched

DaLiV, 2022-10-26:
+ Win version of compilation - same F_CPU issue
compilation command :
D:\3D\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware D:\3D\arduino-nightly\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -tools D:\3D\arduino-nightly\tools-builder -tools D:\3D\arduino-nightly\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries D:\3D\arduino-nightly\libraries -libraries D:\Users\User\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF4:pnum=GENERIC_F429ZGTX,upload_method=hidMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,dbg=none,rtlib=nano -vid-pid=0483_5740 -ide-version=10820 -build-path t:\Temp -warnings=default -build-cache D:\Temp\arduino_cache_572847 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.STM32Tools.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -verbose D:\3D\Marlin\Marlin-BTT\Marlin\Marlin.ino
D:\3D\arduino-nightly\arduino-builder -compile -logger=machine -hardware D:\3D\arduino-nightly\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -tools D:\3D\arduino-nightly\tools-builder -tools D:\3D\arduino-nightly\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries D:\3D\arduino-nightly\libraries -libraries D:\Users\User\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF4:pnum=GENERIC_F429ZGTX,upload_method=hidMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,dbg=none,rtlib=nano -vid-pid=0483_5740 -ide-version=10820 -build-path t:\Temp -warnings=default -build-cache D:\Temp\arduino_cache_572847 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.STM32Tools.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -verbose D:\3D\Marlin\Marlin-BTT\Marlin\Marlin.ino

Result:

In file included from D:\3D\Marlin\Marlin-BTT\Marlin\src\MarlinCore.cpp:33:
D:\3D\Marlin\Marlin-BTT\Marlin\src\MarlinCore.cpp: In function 'void minkill(bool)':
D:\3D\Marlin\Marlin-BTT\Marlin\src\HAL\shared\Delay.h:103:50: error: expected primary-expression before ')' token
  103 |   #define DELAY_US(x) DelayCycleFnc((x) * ((F_CPU) / 1000000UL))
      |                                                  ^
D:\3D\Marlin\Marlin-BTT\Marlin\src\MarlinCore.cpp:920:28: note: in expansion of macro 'DELAY_US'
  920 |   for (int i = 1000; i--;) DELAY_US(600);
      |                            ^~~~~~~~
D:\3D\Marlin\Marlin-BTT\Marlin\src\HAL\shared\Delay.h:103:50: error: expected primary-expression before ')' token
  103 |   #define DELAY_US(x) DelayCycleFnc((x) * ((F_CPU) / 1000000UL))
      |                                                  ^
D:\3D\Marlin\Marlin-BTT\Marlin\src\MarlinCore.cpp:925:28: note: in expansion of macro 'DELAY_US'
  925 |   for (int i = 1000; i--;) DELAY_US(250);
      |                            ^~~~~~~~

GrossoRoberto, 2023-02-05:
Hi did you get some new result ?  i'm in the same point

DaLiV, 2023-02-06:
i'm ended up onto platformio (also moved all compilations to the linux box) ... 
+ can say - big plus on "pio" is next : no each time full recompile for every support library