bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #139: missing documentation for Pro variant

power-max, 2022-12-18:
The arrangement of the connectors and the pinouts are different between the base model and the pro model. most of the documentation on this repository appears to be for the base model and not the pro model, making it difficult to identify what is what on the pro model and forcing me to make some assumptions of how things were shuffled about.

power-max, 2022-12-18:
I'm dumb and I didn't see there indeed was a separate repository for the pro variant! 🤦 A link to it in the this vastly more popular repo would be helpful, since [this](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro) one gets lost in the noise.

maxholgasson, 2023-02-22:
Thanks I was looking for the same. Indeen the documentation is so important for a product and hope it will be improved soon