bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #141: Problems with Octopus v1.1

ctra0031, 2023-01-12:
Hi, I have several problems going on with the Octopus v1.1 board:

1. When I try to restart the board with a new firmware.bin by pressing the restart button, nothing happens.

- I tried powering up the board with both external power supply and via the USB-C port.
- The LEDs indicating board power (top left red LED) and microSD Card plugged in (right blue LED) are on with no flashing.
- I tried with both 32GB and 16GB microSD (FAT32) but there's no difference.

2. When I connect the board with a TFT50 v3.0.1 LCD Screen, it works normally with the reset wire not plugged in, but once I use the reset wire, the whole screen black out.

- With 4 wires plugged in, only the GND -> Reset on the TFT side shorts, with 5 wires, the GND -> Reset on the Octopus side shorts as well.
- I notice that there's a continuity on GND -> 5V (and the other pins -> 5V as well), but not in reverse, not sure if this is a diode behaviour or there's something wrong.

3. When I connect with a stepper motor, the motor is locked and I can't rotate it. I've tried manually turning the shaft, using the LCD board, and on pronterface, but no luck at all.

All of these has been tested on another Octopus v1.1 board (which is not available to me) and no problem occurs. Is there a bug I can fix by myself or is it simply a broken board? Any help would be much appreciated. Thanks.

aingez, 2023-05-22:
Hi, is there any solution . I'm having the same problem as your first one.

Arnie3d, 2023-12-12:
I have the same and it doesn't look like there are any solutions....why ?

bigblocknova454, 2023-12-26:
exact same issue. connect the BTT mini 12864 display board wont boot/install firmware/enter dfu mode nada. unplug display works fine for all the things. Cannot for the life of me figure out what is causing this.

andrew-manger, 2025-11-04:
Still nothing on this?