bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #146: Can't establish a serial communication

lucaguida39, 2023-03-31:
Hi, i'm trying to establish a serial communication with the board but, when i connect to my computer using the usb-c cable, the board is not seen. (I use arduino IDE to send commands and no port is shown).
The firmware is Marlin and i have this settings for serial communication:

#define SERIAL_PORT -1 
#define BAUDRATE 115200

#define SERIAL_PORT_2 1 
#define BAUDRATE_2 115200

How can i solve the problem?

cameronsocialhardware, 2023-06-01:
Did you solve this issue as I'm also facing the same problem. Thanks.

Lorenzo1205, 2023-06-19:
having the same issue ... GRRRRrrr
Now about one week constantly compiling en testing...
Nothing works....

lucaguida39, 2023-07-05:
> 

Didn't solve, I'm sorry