bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #151: Stock firmware for octopus pro v1.1 does not compile

ElfyShort, 2023-07-14:
I am trying to compile stock firmware for octopus pro v1.1 in Platformio.
Getting the error:
'PinMap_SD' was not declared in this scope; did you mean 'PinMap_USB'?

Clean Marlin 2.1.2.1 compiles on my setup with no problem.

Any help will be appreciated!

Here is the full build log:

Compiling .pio\build\STM32H723Zx_btt\src\src\HAL\STM32\sdio.cpp.o
Marlin\src\HAL\STM32\sdio.cpp: In function 'void HAL_SD_MspInit(SD_HandleTypeDef*)':
Marlin\src\HAL\STM32\sdio.cpp:91:26: error: 'PinMap_SD' was not declared in this scope; did you mean 'PinMap_USB'?
   91 |     pinmap_pinout(PC_12, PinMap_SD);
      |                          ^~~~~~~~~
      |                          PinMap_USB
Compiling .pio\build\STM32H723Zx_btt\src\src\inc\Warnings.cpp.o
Marlin\src\inc\Warnings.cpp:724:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  724 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
*** [.pio\build\STM32H723Zx_btt\src\src\HAL\STM32\sdio.cpp.o] Error 1

ruturajshinde, 2023-11-04:
hey, @ElfyShort, were you able to find a solution for this one. I'm facing similar issue and need to compile it for one of my machine. based of 723H

dblatt87, 2024-01-03:
Same problem here, is there a known workaround?

dblatt87, 2024-01-08:
Hello
I try to use a BTT Octopus Pro V1.1 with the STM32H723Z.
I spend hours in back-tracing and also other platform versions in the STM32H723Zx.ini but unfortunately I'm shot out of ideas.
Any help would be very appreciated.

Here the build logs, normal and in verbose mode:
[build_log.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/13861121/build_log.txt)
[build_log-v.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/files/13861123/build_log-v.txt)

Developer-Expsol, 2024-09-29:
Hello
I got a workaround this error.
You have to add the below line
extern const PinMap PinMap_SD[];
after "//*** SD ***", in the following file
.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\inc\PeripheralPins.h

After this change, the firmware get compiled without error, but there is still one problem, the onbaord SD card adapter does not read SD card.

I was able to update the firmware using onboard SD card adapter, but cannot read gcode files from it.
When I use M20, I get the below response

12:34:23.165 : N19 M20*41
12:34:23.165 : echo:No media