bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #153: No connection over usb-c ant uart

Afrem444, 2023-09-27:
I have the problem that i want to set up an octopus v1.1 with klipper and I wont get a response from the board. When Inster the firmware.bin with the micro-sd  and take it out again the file changed to FIRMWARE.cur.

Please help Marlon

Drake143Jack, 2023-10-04:
same issue here
firmware changes to FIRMWARE.cur but octoprint reports "mcu 'mcu': Unable to connect"
tried DFU mode flashing too, but didn't help either

PZrk73, 2023-10-04:
In most cases you was not correctly defined port in configuration of klipper. Or it was not set correctly in the firmware settings.
In firmware it must "Communication interface (USB (on PA11/PA12))" for usb-c for STM32F429.
In config printer.cfg for my system it "serial: /dev/serial/by-id/usb-Klipper_stm32f429xx_XXXXXXXXXXXXXXXXXXXXXXXX-XXXX"
FIRMWARE.cut -- it means all right flashed.

Drake143Jack, 2023-10-04:
lsusb correctly shows "Bus 001 Device 003: ID 1d50:614e OpenMoko, Inc. stm32f446xx"
but "ls /dev/serial/by-id/*" says "ls: cannot access '/dev/serial/by-id/*': No such file or directory"

PZrk73, 2023-10-04:
look at "ls /dev" with and without a connected device and compare. maybe there will be. when connected as COM, it was defined as /dev/ttyS0 or something like that...

Drake143Jack, 2023-10-04:
> look at "ls /dev" with and without a connected device and compare. maybe there will be. when connected as COM, it was defined as /dev/ttyS0 or something like that...

yes /dev/ttyACM0 showed up 
added that instead to klipper printer config serial port 

and the solution for me was that i had added the wrong user to the dialout and tty groups and hence weren't able to access the serial

thanks for your help

PZrk73, 2023-10-04:
> yes /dev/ttyACM0 showed up added that instead to klipper printer config serial port
>
> and the solution for me was that i had added the wrong user to the dialout and tty groups and hence weren't able [...]

Oh, yeah... I forgot about the rights to dialout. I have already encountered this, but I forgot. But I was glad to help find the right way to solve the problem.