bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #157: Robotics

YeetParry, 2023-10-20:
I am trying to use my V1.1 bot for a robotics project, what should I do in order to install the firmware into the board? Thanks

PZrk73, 2023-10-21:
> I am trying to use my V1.1 bot for a robotics project, what should I do in order to install the firmware into the board? Thanks

The same as for the 3D-printer: rename you firmware file to "firmware.bin", write it to SD-card, put it SD-card to slot on your board and press reset button. LEDs blink few times and after a few seconds all be done.

YeetParry, 2023-10-23:
What environment can I write the code in? I am not planning to use
Marlin/Klipper for this project, does the board support custom codes?
Thanks

JannikWilhelm, 2024-12-08:
I recently found a way to program it with arduino-code using PlatformIO:
https://github.com/maxgerhardt/nucleo-f446ze-with-arduino