bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #165: Driver drv8825 plugged in the wrong way, board defective?

Skippydi, 2024-04-22:
I accidentally plugged the drv8825 driver into my Octopus V1.1 board the wrong way round. Now the board is neither recognized via USB nor can I flash it. What could be wrong? Shouldn't the board be able to withstand something like that?

Daneild, 2024-05-07:
I have the same board, there are some fuses on it, check them.  Also, try and use the jumper to be powered on from usb and do not connect the power to it. Hope it will help..somehow.

looxonline, 2024-05-13:
Unfortunately this sort of error connection is difficult to protect against and does indeed cause damage to the motherboard. We included protection against in in earlier motherboards but it unfortunately can cause other issues due to increased resistance in the ground path. Our solution was to introduce the EZ range of drivers which completely prevent incorrectly inserting the drivers by keying them. You will struggle to find any motherboard on the market that uses stepstick drivers and can effectively protect against reverse insertion.

closed 2024-05-13 by looxonline